BelfrySCAD/BOSL2 #731: [BUG]: half shows allways half from wrong side

3Dietrich, 2021-11-16:
**Describe the bug**
I know that the answer is allways "a problem with convexity", but with `.._half()` it is kind of **_always_** and annoying. It's a lot of work to go through all possible 3D objects that _could_ be the reason and in most cases there is no chang. In fact the function is beautyful in theory (really!) but I can't use it anymore because it never shows the half I want to look at.

**Code To Reproduce Bug**
Just an sphere() - and half doesn't show the half:
```
include <BOSL2/std.scad>
front_half()
sphere(33);
```
Sphere doesn't even support 'convexity'.. so it has to be corrected in the half code..

**Expected behavior**
It should show the half of the sphere/object, from all sites
**Screenshots**
https://user-images.githubusercontent.com/4608796/142082028-999a7328-40b9-4487-881a-65cb7c0e9eaa.mov

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2021.01
- Other libraries used: no

revarbat, 2021-11-17:
This is a known bug either in OpenSCAD, or in the OpenCSG display library it uses.
You may be able to work around it by passing a smaller value for `s=` to `front_half()`, like `front_half(s=100) ...`

revarbat, 2021-11-17:
Oh, and it has nothing to do with convexity, for once.  :smile:

adrianVmariano, 2021-11-17:
If you render your object then it will work correctly.  The problem is that you need the cube that cuts away the object to basically be in the viewport, so if it's too big then the preview algorithm fails.  As Revar says, using the s parameter should fix it....if you can figure out the right s parameter value to use.

adrianVmariano, 2021-11-17:
It seems like the issue with preview failing when s is large should be more clear in the docs.  Also another possible solution here might be to remove the default value for s and force the user to pick some value.

adrianVmariano, 2021-11-20:
We have reduced s to 100 and I just made some further clarifications to the manual.  Is there anything else we can do on this issue?

3Dietrich, 2021-11-20:
super! Thank you

closed 2021-11-20 by 3Dietrich