BelfrySCAD/BOSL2 #732: cuboid roundings incorrect of $fn not divisible by 4

adrianVmariano, 2021-11-17:
Rounding some (not all) edges gives incorrect results:

    cuboid(10, rounding=2, edges=TOP,$fn=5);
[image]

    cuboid(10, rounding=2, edges=TOP,$fn=25);

[image]

revarbat, 2022-01-13:
I think the only way to fix this is to quantize $fn to 4 inside the internal module `corner_shape()`.

revarbat, 2022-01-13:
Well, it may be possible to carefully align all the sub-cylinders, but I think the results will still look kind of bad compared to just quantizing $fn to 4.

adrianVmariano, 2022-01-13:
I think forcing $fn up to the next multiple of 4 is a totally reasonable solution.

adrianVmariano, 2022-01-13:
Consistent with what?

revarbat, 2022-01-13:
Consistent in that both `xcyl()` and `ycyl()` should have an edge facing the same way (up or down).  I assume both `cyl()` and `zcyl()` and `cylinder()` have an edge towards X+.

adrianVmariano, 2022-01-13:
Do zcyl and cylinder have an edge towards x+ or a line towards x+?

revarbat, 2022-01-13:
`xcyl()` has an edge towards Z- and a face towards Z+:
[image]
`ycyl()` has an edge towards X+ and a face towards X-:
[image]
`zcyl()` has an edge towards X+ as well, and a face towards X-:
[image]

revarbat, 2022-01-13:
`cyl()` and `cylinder()` are oriented like `zcyl()`.

revarbat, 2022-01-13:
I'm thinking both `xcyl()` and `ycyl()` should have their starting edge towards Z+

revarbat, 2022-01-13:
`cuboid()` is fixed by quantization in PR #755

adrianVmariano, 2022-01-13:
If cylinders have starting edge towards z+ then they'll have an edge down for even $fn.  Is that ok?

revarbat, 2022-01-20:
Original issue fixed.

closed 2022-01-20 by revarbat