BelfrySCAD/BOSL2 #733: rect() should support negative roundings

adrianVmariano, 2021-11-17:
Right now if you give rect() a negative rounding it is simply ignored.  Negative roundings should actually work to give fillets.  (And if they don't it should be an error.)

adrianVmariano, 2022-01-15:
I disagree that this is low priority.  At the very least, you need to have an error for negative roundings.  But pretty much everywhere roundings are accepted, you can use negative ones, so the lack of support for it here is an anomaly.   I tried to use negative roundings on rect() in an actual model, so it's not just arbitrary completism.   Is there some reason not to add this?  (There's the ambiguity about which direction the negative rounds point.)

revarbat, 2022-01-15:
I actually already tried to implement negative rounding/chamfering in `rect()` yesterday. The way it's written now, it's going to need a complete rewrite in how it creates the path. Which is doable. I just had to scrap my first attempt.

adrianVmariano, 2022-01-15:
I was reacting to the "low priority" tag on this issue.

revarbat, 2022-01-20:
Implemented.

closed 2022-01-20 by revarbat