BelfrySCAD/BOSL2 #739: Missing sp_cap() to accompany sp_neck()

pp19dd, 2021-12-06:
Of all the OpenSCAD thread libraries I've seen, this library is the only one that includes a very appropriate `sp_neck()` for making large jar lids (aiming for around 90-100mm diameter).  Thank you!  However, it doesn't include a corresponding sp_cap module that mates with it.  

Would it be possible to add an sp_cap() module to bottlecaps.scad ?

adrianVmariano, 2021-12-08:
The problem is that SPI necks are based on published standards whereas SPI caps seem to be a trade secret.  This kinda thwarted my ability to do the SPI cap.  I was pondering whether to try measuring some caps to get ideas of how it works.  But wasn't sure how to proceed.   It's an obvious need.

adrianVmariano, 2022-10-16:
I made a stab at this.  It's in PR #970.

closed 2022-10-18 by adrianVmariano