BelfrySCAD/BOSL2 #765: Trivial Issue with centering things

Dorro101, 2022-01-21:
Hi Revar

I have a trivial issue with the constant name "CENTER". Is it possible you could add an equivalent constant "CENTRE"? I see you have done this other constants like "BOT" and "BOTTOM"? I cannot help typing CENTRE and its drives me a little batty, especially late at night after a long coding session. I edit a lot, use UK/South African spelling, so it throws me a little.

Thanks for your great libraries. BOSL2's attachments rock and are perfect for my current parametric project!

Ciao
Dorian

revarbat, 2022-01-21:
The error looks like you passed `anchor="CENTER"` (a string) as opposed to `anchor=CENTER` (the presumably added constant.).

revarbat, 2022-01-21:
This was trivial, so I added a CENTRE alias to CENTER.

closed 2022-01-21 by revarbat

Dorro101, 2022-01-21:
Hi Revar

Thanks. I was just being an idiot. I had already declared 'CENTRE' locally as a workaround, but forgot to remove it when I changed the 'constants.scad' file. Thanks for adding, now I won't have to change it myself every time I update the code from Github.

Thanks again for your great work!!!

Ciao
Dorian

Closed #765<https://github.com/revarbat/BOSL2/issues/765>.