BelfrySCAD/BOSL2 #777: tag() inconsistency with $tags=

adrianVmariano, 2022-02-05:
There is an inconsistency between using tag() and settings the tags by other means.  

    tags("foo"){
        tags("bar"){ ... }
    }

will fail to have the desired effect because tags() doesn't run its children if they aren't "shown".  So if the above is under `diff("foo",keep="bar")` then you get a surprise that the "bar" tagged item is not kept. 

The question then is: is there ever some situation where this aggressive truncation of the tree is actually desirable?  Or should tags() be changed to run its children?  

In pondering the example of nesting from issue #617 it seems that the problem is knowing where to give the tags.  So there the example:

    diff("foo")
    cube(100){
      attach(CENTER)
         diff("bar")
         cyl(d=70,h=100,anchor=BOTTOM,$tags="foo")
           cyl(d=20,h=200,$tags="bar");
    }

this example fails.  But the problem is placement of the tag.  So it will work like this:

    diff("foo")
    cube(100){
      attach(CENTER)
         diff("bar",$tags="foo")
         cyl(d=70,h=100,anchor=BOTTOM)
           cyl(d=20,h=200,$tags="bar");
    }   

The main point here is that it wasn't the use of tags() that was the key to success, but a scoping issue for the tag.  So again I ask, is there ever a situation where the current behavior of tags() is the right behavior?

adrianVmariano, 2022-03-16:
Is the conclusion here that we will change `tags` to just set the variable and add some other `apply_tags` module that force applies the tags to the children so it can support non-attachable modules?

revarbat, 2022-03-17:
This is the consensus that I recall, yes.

closed 2022-03-20 by revarbat