BelfrySCAD/BOSL2 #780: anchors for rounded rectangles and position relative to edges

adrianVmariano, 2022-02-09:
If you create a rounded rectangle with `rect(size,R=r)` then it is impossible to position it relative to its edges, for example to place it in the first quadrant with one edge on each axis.  This is because the LEFT+FRONT anchor is on the edge of the roundover instead of on the imaginary corner--the bounding box corner.  So the analogous `square()` invocation produces a square in the first quadrant (without rounding) and getting the matching rounded result is impossible (without shifting it manually into position by `R*(1-1/sqrt(2))`).

It seems like there are many situations where the bounding box corners are more useful and more intuitive anchor points than the actual anchors.   So I wonder if there should be a flag on `rect()` to switch between bounding box and actual surface anchors.  For example `box_anchors=true`.  Or maybe bounding box anchors should be the default and then you have `surf_anchors=true` to select anchors that are actually on the rounding.   I guess another option would be something like `atype="box"` which sort of parallels `cp="box"`.  

If something like this is implemented it probably needs to go on cuboid and maybe some other functions that also do roundings.

revarbat, 2022-03-09:
`rect()` has been changed to default to anchoring to the bounding box, with `atype="perim"` for anchoring to the rounded perimeter.

closed 2022-03-09 by revarbat

adrianVmariano, 2022-03-09:
It occurs to me that "perim" doesn't generalized to 3d.