BelfrySCAD/BOSL2 #781: absolute direction (turn angle) for turtle

carstenfuchs, 2022-02-10:
**Is your feature request related to a problem? Please describe.**
The 2D turtle can only make relative turns, but the direction cannot be set or reset to an absolute value, e.g. 0°, 270°, …
This is useful after a turn to an odd angle that we want to recover from, i.e. for continuing in a well known direction.

**Describe the solution you'd like**
Another command like e.g. "turn_absolute" or "turn_compass_heading" that sets the direction of the turtle wrt. the global coordinate system.
Or simply "reset_turn" to go back to direction 0°.

**Describe alternatives you've considered**
See example code below.

**Example Code**
```
// Code goes here.
w = 31.415926;
path = turtle([
    "turn", 90,
    "move", 10,
    "turn", w,
    "turn", -w + 90,   // An alternative / work-around for:  "turn_abs", 180
]);
```

**Additional context**
Absolute values are already possible for positions, but not for turn angles.

adrianVmariano, 2022-02-10:
The requested feature already exists under the name "setdir" for "set direction".

carstenfuchs, 2022-02-10:
Oh. This is very embarrassing! I scanned the list to check if the feature was already there, but I must have been blind.
Sorry for the noise!

closed 2022-02-10 by carstenfuchs