BelfrySCAD/BOSL2 #783: vnf_bend around X axis

bob-ewart, 2022-02-12:
The example in the documentation with the horizontal cube raised up and bent around the "Y" axis works.
When I try to bend a vertical cube around the "X" axis it fails.
    vnfx0 = cube([40,10,100]);
    vnfx1 = up(1,p=fwd(50, p=vnfx0));
    #vnf_polyhedron(vnfx1)
     bentx1 = vnf_bend(vnfx1, axis="X");
     vnf_polyhedron(bentx1);
I get:
[ERROR: Assertion 'is_vector(cuts)' failed: "The split list must be a vector." in file ../.local/share/OpenSCAD/libraries/BOSL2/vnf.scad, line 744](https://github.com/revarbat/BOSL2/issues/744,/home/bob/.local/share/OpenSCAD/libraries/BOSL2/vnf.scad)
[TRACE: called by '_slice_3dpolygons' in file ../.local/share/OpenSCAD/libraries/BOSL2/vnf.scad, line 688](https://github.com/revarbat/BOSL2/issues/688,/home/bob/.local/share/OpenSCAD/libraries/BOSL2/vnf.scad)
[TRACE: called by 'vnf_slice' in file ../.local/share/OpenSCAD/libraries/BOSL2/vnf.scad, line 1139](https://github.com/revarbat/BOSL2/issues/1139,/home/bob/.local/share/OpenSCAD/libraries/BOSL2/vnf.scad)

When I change the second line to:
   vnfx1 = fwd(50, p=vnfx0);
I get:
[ERROR: Assertion '((bmin.z > 0) || (bmax.z < 0))' failed: "Entire shape MUST be completely above or below z=0." in file ../.local/share/OpenSCAD/libraries/BOSL2/vnf.scad, line 1134](https://github.com/revarbat/BOSL2/issues/1134,/home/bob/.local/share/OpenSCAD/libraries/BOSL2/vnf.scad)
[TRACE: called by 'vnf_bend' in file testVNFbend.scad, line 18](https://github.com/revarbat/BOSL2/issues/18,/home/bob/tmp/testVNFbend.scad)

bob-ewart, 2022-02-12:
That's with OpenSCAD 2022.02.11 nightly
with OpenSCAD 2019.05 it gets:
ERROR: Parser error in file "/home/bob/.local/share/OpenSCAD/libraries/BOSL2/geometry.scad", line 961: syntax error 
because of the use of the ^ operator in the BOSL2 library dated 2022-02-08 7:07 PM

I get similar errors to the last message with OpenSCAD 2019-05 using the BOSL2 library dated 2021-10-17

adrianVmariano, 2022-02-13:
What is happening is that you have got a weirdly tall object and narrow object to bend around the x axis.   You need to pick an extremely large $fn for this to work because you have such a tiny angle, or it ends up choosing to use zero steps.  Clearly there should be a check for this case so it doesn't bomb, but it's a sort of useless case.  You can make it work right now by simply choosing a sufficiently large $fn or small enough $fs/$fa.  Looks like $fn=64 is the minimum, which I think gives you one step.  It's such a small angle that one step is really enough.  

I might suggest that if you're going to wrap an object around a cylinder, the object should be sort of thin in the z direction and large in the other directions.

adrianVmariano, 2022-02-14:
So Bob, is this working for you at this point?

bob-ewart, 2022-02-15:
Yes, sort of

I did switch to the AppImage version 2021.01 and built a small test case using a hollow cylinder with a lip on one end.  It works and renders.  

If I add another unconnected cube previewing is OK.  
Rendering gives ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. 
It also renders OK without the lip and with the cube.

include <BOSL2/std.scad>

add_block   = false;
add_lip     = true;
$fn         = 64;
bend        = 30;

echo("\n\n--Test vnf_bend with cube\n");

shape   = difference(circle(d=15),circle(d=5));
lip     = up(50-.75, p=path_sweep(circle(d=1.5), circle(d=15), closed=true));

radius  = 50 * 360 / (2 * PI * bend);
echo(radius = radius);
vnf1    = linear_sweep(shape,50);
vnf2    = (add_lip)?
            vnf_join([vnf1, lip]):
            vnf1;
vnf3    = xrot(-90, p=vnf2);
vnf4    = up(radius, p=vnf3);
vnf5    = vnf_bend(vnf4, axis="X");
back(radius) xrot(90) vnf_polyhedron(vnf5);

if(add_block) {
    left(20) cube(10);
}

adrianVmariano, 2022-02-15:
So it's important to remember that a single shape always works.  Adding the cube tells you if it's actually OK.

The problem with your model above is that you are using vnf_join incorrectly.  It is meant to be used to join partial polyhedra, like if you have half a shape and then you build the other half.  You are trying to use it to perform a generic union, which produces an invalid result.  You need to separately manipulate the VNF resulting from your lip and the main body and then union those instead of trying to union the VNF.   Or alternatively, use offset_sweep to add the lip in an integrated fashion into a single VNF directly, though getting the hole down the middle then is maybe a little tricky.  

    include <BOSL2-fork/std.scad>

    add_block   = true;
    $fn         = 64;
    bend        = 30;

    echo("\n\n--Test vnf_bend with cube\n");

    shape   = difference(circle(d=15),circle(d=5));
    lip     = up(50-.75, p=path_sweep(circle(d=1.5), circle(d=15), closed=true));

    radius  = 50 * 360 / (2 * PI * bend);
    echo(radius = radius);
    body    = linear_sweep(shape,50);

    T = up(radius)*xrot(-90);

    rad = pointlist_bounds(lip[0])[1].z;   // Need to have common radius

    vnf_polyhedron(vnf_bend(apply(T,lip),r=rad,axis="X"));
    vnf_polyhedron(vnf_bend(apply(T,body),r=rad, axis="X"));

    if(add_block) {
        left(20) cube(10);
    }

The above works.  It's not what I'd call fast.  I wonder if you can apply the lip at the bottom instead of the top and not use "bend" on it.  I don't think the bend is doing much to the lip, and it more than doubles the time.

adrianVmariano, 2022-02-15:
Looks like I screwed up the bend radius computation.  Should be:

    T = up(radius)*xrot(-90);
    rad = pointlist_bounds(apply(T,lip)[0])[1].z; // Need to have common radius

Note that if this is your whole design then you can avoid bend and just use path_sweep with an arc path.  

It does seem like the way vnf_bend is implemented is pretty confusing.  I wonder if it should take an object at the origin and a bend radius instead of trying to infer a bend radius from the object position.  

    lip2 = up(-.75, p=path_sweep(circle(d=1.5), circle(d=15), closed=true));
    bendr = 100;
    length = 50;
    final_angle = length/2/PI/bendr * 360;
    arc = arc(r=bendr, angle = [0, final_angle], $fn=128);
    xrot(90)path_sweep(shape, arc);
    yrot(-final_angle)right(bendr)vnf_polyhedron(lip2);  // could use rotate_extrude here

bob-ewart, 2022-02-16:
In my main module I use something like:
a = [for (i=[0:levels]) up(i*maxZ/levels)*xscale(scaleX[i])* yscale(scaleY[i])];
shank = sweep(shape,a);
Where scaleX and scaleY are arbitrary curves based on measurements from the bit I'm trying to duplicate.

Your idea of taking an object at origin and a bend angle (not radius) makes a lot of sense.  You can calculate the rest.

This is my current test case with an oval shape.  I've tested it up to a bend angle of 270 although 90 would be the max in use.  The red and green marks show the circle of the bend and the position of the lip.

include <BOSL2/std.scad>
echo("\n\n---oval test----------\n");
maxZ    = 60;   // height of stem
bend    = 30;   // bend of shank
shankX  = 13;   // x diameter of shank oval
shankY  =  9;   // y diameter of shank oval
hole    =  6;   // interior hole diameter
lipD    = 1.5;  // diameter of lip bead
showMarks =  true;

$fn     = 64;

shape   = yscale(shankY/shankX, p= difference(circle(d=shankX), circle(d=hole)));
echo("shape bounds", pointlist_bounds(shape[0]));
radius  = maxZ * 360 / (2 * PI * bend) - pointlist_bounds(shape[0])[1][1];

if(showMarks) {
    xrot(-90) yrot(-90) right(radius) color("green") torus(radius,1);
    color("red") back(radius) xrot(90 - bend) cube([1,1,radius+shankY]);
}

shank   = linear_sweep(shape, maxZ);
vnf1    = xrot(-90, p=shank);
vnf2    = up(radius, p= vnf1);
stem    = vnf_bend(vnf2, axis="X");
back(radius) xrot(90) vnf_polyhedron(stem);

lip     = yscale(shankY/shankX, p=torus(r_maj=shankX/2,r_min=lipD/2, center=true));
back(radius) xrot(90-bend) up(radius) xrot(90) up(lipD/2) vnf_polyhedron(lip);

adrianVmariano, 2022-02-16:
No, "bend angle" is not a well defined concept, and then it depends on the extent of the shape, which is what complicated things for aligning the lip and the tube.  Using radius is simpler and more clear, and makes it possible to bend multiple separate objects in a compatible way, or in a way where you know the relation between the bends.  The main complication of my idea is that a directional selector is needed for bend in the positive or negative direction.  

Your version with an oval shape doesn't change anything about my suggestion to avoid vnf_bend and just construct the curve directly.  It is much faster.

bob-ewart, 2022-02-16:
In my main module I use something like:
a = [for (i=[0:levels]) up(i*maxZ/levels)*xscale(scaleX[i])* yscale(scaleY[i])];
shank = sweep(shape,a);
Where scaleX and scaleY are arbitrary curves based on measurements from the bit I'm trying to duplicate.

Your idea of taking an object at origin and a bend angle (not radius) makes a lot of sense.  You can calculate the rest.

This is my current test case with an oval shape.  I've tested it up to a bend angle of 270 although 90 would be the max in use.  The red, blue and green marks show the circle of the bend and the position of the lip.

I edited this to show what happens when the shank is moved up or down.

include <BOSL2/std.scad>
echo("\n\n---oval test----------\n");
bend    = 30;   // bend of shank
z_offset=  0;   // shift the shank up or down 
showMarks =  true;
module dummy(){};
function arcLength(r, theta) = 2*PI*r * theta/360;
function arcRadius(arcLength, theta) = 360 *  arcLength /(2 * PI * theta);
function arcAngle(arcLength, r) = arcLength * 360 / (2 * PI * r);

maxZ    = 60;   // height of stem
shankX  = 13;   // x diameter of shank oval
shankY  =  9;   // y diameter of shank oval
hole    =  6;   // interior hole diameter
lipD    = 1.5;  // diameter of lip bead

$fn     = 64;

shape   = yscale(shankY/shankX, p= difference(circle(d=shankX), circle(d=hole)));
radius_adjust = pointlist_bounds(shape[0])[1][1];

radius  = maxZ * 360 / (2 * PI * bend) - radius_adjust;
echo(radius=radius);

if(showMarks) {
    xrot(-90) yrot(-90) right(radius) color("green") torus(radius,1);
    color("red") back(radius) xrot(90 - bend - arcAngle(z_offset,radius+radius_adjust)) cube([1,1,radius+shankY]);
    color("blue") back(radius) xrot(90 - arcAngle(z_offset,radius+radius_adjust)) cube([1,1,radius+shankY]);
}

shank   = up(z_offset, p=linear_sweep(shape, maxZ));
vnf1    = xrot(-90, p=shank);
vnf2    = up(radius, p= vnf1);
stem    = vnf_bend(vnf2, axis="X");
back(radius) xrot(90) vnf_polyhedron(stem);

lip     = yscale(shankY/shankX, p=torus(r_maj=shankX/2,r_min=lipD/2, center=true));
back(radius) xrot(90-bend-arcAngle(z_offset,radius+radius_adjust)) up(radius) xrot(90) up(lipD/2) vnf_polyhedron(lip);

adrianVmariano, 2022-02-16:
I'm not sure what you're point is in your last message.  If I change z_offset it moves up and down.  The result is exactly as expected.

bob-ewart, 2022-02-16:
If the top and bottom planes of the object are parallel, I want to bend the object so that the angle between the planes is given.  Is the radius going to the outside, center or inside.  In any event I would have to calculate it.

My previous example was just to show that the object could be placed above, below or crossing z=0;

The oval case was just a simplification to do the bend and place the lip.  I use sweep() because I don't think the others allow for complex separate scaling in X and Y.
Here is a simplified version of what I'm trying to do:

include <BOSL2/std.scad>
echo("\n\n---real test----------\n");
bend    = 30;   // bend of shank
showMarks =  false;
module dummy(){};

maxZ    = 58;   // height of stem
baseD   = 13;   // diameter of base
hole    =  6;   // interior hole diameter
lipD    =  1.5; // diameter of lip bead
levels  = 15;   // levels for sweep 

scaleX = [1, 1.01602, 1.01107, 1.00454, 1.00807, 1.02217, 1.04836, 1.07545, 1.10872, 1.14227, 1.175, 1.20638, 1.23648, 1.2585, 1.28196, 1.32409];
scaleY = [1, 0.842092, 0.732918, 0.677169, 0.635601, 0.604351, 0.574996, 0.545409, 0.51434, 0.48284, 0.446754, 0.408295, 0.367952, 0.326629, 0.285803, 0.218978];

a = [for (i=[0:levels]) up(i*maxZ/levels)*xscale(scaleX[i])* yscale(scaleY[i])];
$fn     = 64;

shape   = difference(circle(d=baseD), circle(d=hole));
radius_adjust = pointlist_bounds(shape[0])[1][1];
radius  = maxZ * 360 / (2 * PI * bend) - radius_adjust;
echo(radius=radius);

if(showMarks) {
    xrot(-90) yrot(-90) right(radius) color("green") torus(radius,1);
    color("red") back(radius) xrot(90 - bend) cube([1,1,radius+baseD]);
}

shank   = sweep(shape, a);
vnf1    = xrot(-90, p=shank);
vnf2    = up(radius, p= vnf1);
stem    = vnf_bend(vnf2, axis="X");
back(radius) xrot(90) vnf_polyhedron(stem);

lipPath = scale([scaleX[levels], scaleY[levels]], p=circle(d=baseD));
lip    = path_sweep(circle(d=lipD), lipPath, closed=true);
back(radius) xrot(90-bend) up(radius) xrot(90) up(lipD/2) vnf_polyhedron(lip);

adrianVmariano, 2022-02-16:
The assumption that the model being bent has a top and bottom parallel to the xy plane is a bad assumption.  No reason to think that should be true.  In the examples on the Wiki things like arrows are bent.  Also there is a spiral that would require a bend angle of 1200 degrees or something.  The current code uses the largest z value to define the radius.  I think this is confusing.  My new idea is that the radius is what you specify and it's based at the appropriate axis.  I think that should be a more natural approach.  

When posting code here, indent your entire code block 4 spaces.  It will then show up as code.  When you post it as text, the markdown interpreter can eat the the asterisks---your code needed fixing before I could run it.  

The fact that you're already using sweep() makes it a trivial change to insert the bend into your sweep transform list.  I think that's the way you should make this model.  

    include <BOSL2-fork/std.scad>

    echo("\n\n---real test----------\n");
    bend = 30; // bend of shank
    showMarks = false;
    module dummy(){};

    maxZ = 58; // height of stem
    baseD = 13; // diameter of base
    hole = 6; // interior hole diameter
    lipD = 1.5; // diameter of lip bead
    levels = 15; // levels for sweep

    scaleX = [1, 1.01602, 1.01107, 1.00454, 1.00807, 1.02217, 1.04836, 1.07545, 1.10872, 1.14227, 1.175, 1.20638, 1.23648, 1.2585, 1.28196, 1.32409];
    scaleY = [1, 0.842092, 0.732918, 0.677169, 0.635601, 0.604351, 0.574996, 0.545409, 0.51434, 0.48284, 0.446754, 0.408295, 0.367952, 0.326629, 0.285803, 0.218978];

    a = [for (i=[0:levels]) xrot(-bend*i/levels,cp=[0,100,0])*/*up(i*maxZ/levels)**/xscale(scaleX[i])*yscale(scaleY[i])];
    $fn = 64;

    shape = difference(circle(d=baseD), circle(d=hole));
    radius_adjust = pointlist_bounds(shape[0])[1][1];
    radius = maxZ * 360 / (2 * PI * bend) - radius_adjust;
    echo(radius=radius);

    if(showMarks) {
    xrot(-90) yrot(-90) right(radius) color("green") torus(radius,1);
    color("red") back(radius) xrot(90 - bend) cube([1,1,radius+baseD]);
    }

    shank = sweep(shape, a);
    vnf1 = xrot(-90, p=shank);
    vnf2 = up(radius, p= vnf1);
    stem = vnf2;//vnf_bend(vnf2, axis="X");
    back(radius) xrot(90) vnf_polyhedron(stem);

    lipPath = scale([scaleX[levels], scaleY[levels]], p=circle(d=baseD));
    lip = path_sweep(circle(d=lipD), lipPath, closed=true);
    xrot(-bend,cp=[0,100,0])vnf_polyhedron(lip);

There's a quick mod, so some things might not be right for your dimensions.  I get this feeling that think this approach doesn't work because your model is somehow "too complicated" but all the complications are independent of this aspect of the model.  Using vnf_bend means that the model has to be sliced into layers and each layer then gets a rotation transformation applied.  Note how very similar this is to what sweep is doing---except you give it the layers up front, so it's much faster.

bob-ewart, 2022-02-17:
Your new idea sounds good, but I would like examples of determining the radius.
Your code is something that needs to be documented in the wiki description of sweep.  

However, it doesn't really work.  It elongates the shape.  Compare bend=20 to bend=60.  The length around the bend should be the same as the unbent shape.

Add at the end of your code to see the difference in lengths:
    b = [for (i=[0:levels]) up(i*maxZ/levels) * xscale(scaleX[i]) * yscale(scaleY[i])];   
    left(20) sweep(shape,b);

A nit, but the lip needs to be "down" by lipD/2
    lip = down(lipD/2, p= path_sweep(circle(d=lipD), lipPath, closed=true));

Sorry about the code, I'm learning.  I should have looked at the preview tab.

adrianVmariano, 2022-02-17:
As I said, I didn't try to perfect this alternate approach, just show the basic idea.  You have full control over the length because you are constructing the shape.  You want it shorter?  Adjust the sweep transforms to make it shorter.  Presumably the radius, which I hard coded at 100, needs to be different depending on the bend angle, and you need to compute it from the angle and desired length.

bob-ewart, 2022-02-17:
Using the computed _radius_ did the trick.  
This exercise has been very interesting and I learned a lot.  I never would have guessed from the documentation that you could use sweep transformations to do the bend.

It's gone from ~3/4 hour using my original brute force to seconds using sweep.

Please keep working on your ideas for vnf_bend and the documentation for it and sweep.

Thank you very much for your help and for teaching me more about manipulating VNFs.

closed 2022-02-17 by bob-ewart