BelfrySCAD/BOSL2 #802: Mutators tutorial doesn't give rotate_extrude example (and doc missing from mutators.scad wiki)

ochafik, 2022-03-14:
The `rotate_extrude` example given in the [mutators tutorial](https://github.com/revarbat/BOSL2/wiki/Tutorial-Mutators#extrusion-mutators) is actually a `linear_extrude`.

Also none of them are documented in https://github.com/revarbat/BOSL2/wiki/mutators.scad :-)

adrianVmariano, 2022-03-14:
rotate_extrude is a regular OpenSCAD module, not a BOSL2 module.  Though clearly there's a mistake in the tutorial there, I think rotate_extrude was mentioned only to lead up to cylindrical_extrude.

adrianVmariano, 2022-03-14:
Maybe the tutorial needs a link to the OpenSCAD manual for those functions?

ochafik, 2022-03-15:
Yes that would be good. Given how many of the stock modules are overridden, the BOSL2 doc & cheatsheet are now my only go to reference

adrianVmariano, 2022-03-15:
I wonder if it makes sense to include at least minimal docs for base openscad modules in the BOSL2 docs.

ochafik, 2022-03-16:
As far as I'm concerned, a quick reminder of the signature + a very simple example (& link to the fuller official doc) would help.

RAMilewski, 2023-03-12:
I can't imagine doing anything in OpenSCAD without BOSL2.   I added an issue in https://github.com/Antyos/vscode-openscad/issues/38 asking for a 1-click open of the BOSL2 cheatsheet  in the vscode-openscad extension to VSCode as it does with the OpenSCAD cheatsheet   ...we'll see what happens.  

The big problem with the OpenSCAD docs is that they're in WikiBooks and the search box there does a search of ALL of Wikibooks which is pretty much useless.  So the cheatsheet is about the only useful link into them.

closed 2023-08-19 by revarbat