BelfrySCAD/BOSL2 #803: Bezier curves in turtle3d?

ochafik, 2022-03-14:
I love how turtle3d is able to create arcs, but find it quite hard to connect to anything once I start rotating (total newbie here, hi!).

Would it be possible / would it make sense to introduce bezier curve commands to the turtle?

Not sure what would be the best way to achieve that, but it seems that control points could be inferred in the simple case (using current direction + move distance, and reverse of that at the end point).

Simple case: infer control points:
```js
"bezierstart",
// first control point inferred to where "move" would have gotten us here
"up", 45, "move", 10, // end point
// second control point inferred to ["move", "reverse"] *after* the turtle commands to navigate to the end point were executed
"bezierend",
```

Manual case: explicit control points (not limited to 2):
```js
"bezierstart",
"up", 45, "move", 10, // 1st ctl point
"left", 45, "move", 10, // 2nd ctl point
...
"up", 45, "move", 10, // end point
"bezierend",
```

Maybe with a `beziersteps` command too (defaulting to `arcsteps` or to `$fn`, `$fa` and `$fs` calculations to pick how to sample the curve).

This would make it trivial to create curves that go to precise places, as opposed to calculating arc directions with the existing "arc*" commands. It would also have the interesting property of being able to comment the "bezierstart" & "bezierend" commands and still reach the same destinations.

Any thoughts?

adrianVmariano, 2022-03-14:
Not sure I understand.  Are you saying that turtle's output would basically be the same as running bezpath_curve on the turtle output without the bezier command?

ochafik, 2022-03-15:
> Are you saying that turtle's output would basically be the same as running bezpath_curve on the turtle output without the bezier command?

@adrianVmariano I was thinking more that the bezier commands wouldn't change the end point of the turtle, making it easier to understand where you land if/when you want to connect to a precise point (i.e. easy to "bezierfy" a path).

Something like this
```js
pts = bezier_turtle3d([
    "bezierstart",
    "bezierctl", ["move", 2], // "forks" the turtle to define a control point
    "left", 45,
    "move", sqrt(2),
    "bezierctl", ["left", 45, "move", -1],
    "bezierend",
    "move", 1,
]);
```

Would draw this:

[image]

adrianVmariano, 2022-03-15:
I'm open to the idea of a better way to specify curves, but I'm afraid this doesn't seem like it.  It's pretty hard to read that command and figure out what it does, and I can only imagine the questions we'd get about how to make it work.  Another complication is that in 3D you need to track rotation of the turtle, and it's not clear how you capture that here, and the bezier concept introduces a sampling problem for rotation.  So some more powerful curve specification might be easier to do in 2d and hard to generalize to 3d.  

If you want a curve that passes through known points (in 2d or 3d) maybe you could use smooth_path(), which is in fact a way to "bezierfy" a path, based on path_to_bezpath().

closed 2022-03-22 by adrianVmariano