BelfrySCAD/BOSL2 #808: Explanation of threaded_rod outer diameter triplet

alidaf, 2022-03-22:
I'm trying to find the inner diameter of an arbitrary threaded rod but there is no function to return it. I would therefore like to define it but I'm confused by the outer diameter triplet [d_min, d_pitch, d_major]. I've tried a range of different numbers for d_pitch, lying in-between d_min and d_major but I can't work out what it actually does and how to effectively use it. I think I would prefer a function to return iso values for whatever outer diameter I supply.

alidaf, 2022-03-22:
I managed to find the relationships between the parameters here: https://en.wikipedia.org/wiki/ISO_metric_screw_thread. Calculating H is easy and everything else depends on that and the OD.

closed 2022-03-22 by alidaf

TheRealBenForce, 2024-11-17:
Came here for this also.

alidaf, 2024-11-17:
> Came here for this also. @alidaf , do you have anything you can share?

@TheRealBenForce  Not really. I can't remember why I specifically wanted to know the inner diameter but since the function produces ISO dimensions, it can be worked out from the link I gave. I have used this function a lot and so far only ever used the basic functionality by defining the outer diameter, pitch and length.