BelfrySCAD/BOSL2 #816: Add font kerning support for arc_of

alidaf, 2022-03-29:
Feature request.

Currently arc_of to print text considers the global font size and works well for monospaced fonts, but non-monospaced fonts can look strange around skinny letters.

I don't know much about fonts but kerning is used to position characters for a more aesthetic look in desktop packages. I don't know how that could be achieved with the SCAD language so perhaps a user-defined modification table can be applied for making adjustments to specific letters by calling arc_of multiple times to make up the segments of the text.

I think it may be possible to add a check in a module that calls the "select" function to check for the character codes and set the $idx accordingly. No example code I'm afraid since I'm new to SCAD but it is something I'll be looking into.

revarbat, 2022-03-29:
OpenSCAD does not yet provide userspace access to font info like kerning, though there is a fontmetrics feature in beta testing. Without the fontmetrics feature, the only way to handle kerning would be a list of character widths for each font. This, while possible, is annoying to implement, and would become obsolete as soon as the fontmetrics feature is merged into mainline.

Having said all that, there is already a way to have text follow a path, which will take fontmetrics into account once it becomes available:

```openscad
include <BOSL2/std.scad>
path = path3d(arc(100, r=25, angle=[245, 370]));
color("red")stroke(path, width=.3);
path_text(path, "Example text", font="Courier", size=5, lettersize = 5/1.2);
```
[image]

adrianVmariano, 2022-03-29:
Just to be clear, currently with the stable OpenSCAD that doesn't have textmetrics, there is NO INFORMATION about the font size available to path_text.  So no, it doesn't "consider" the font size.  There's no way to do that.  You have to tell it what letter spacing you want with the lettersize arg and that's what you get.  It's totally decoupled from the actual size of the font.  You can set lettersize to an array to individually position all the letters if you want to mess with that, so arguably kerning is available.  

Now arguably with textmetrics turned on you don't have kerning, so maybe lettersize should still be used in that case to perform relative adjustment of the size if you want to tweak the layout from the font metric size?

adrianVmariano, 2022-03-29:
Better to add a new kern parameter to make adjustments when textmetrics are used as opposed to repurposing lettersize, which seems sort of confusing?

adrianVmariano, 2022-03-30:
Actually it looks like the problem the original poster had was using arc_of() instead of path_text().

adrianVmariano, 2022-03-30:
Added a kern parameter to path_text.

closed 2022-03-30 by adrianVmariano

alidaf, 2022-03-30:
Many thanks for looking into this @adrianVmariano . I woke up to quite a few messages this morning. I used the arc_of() because that is the example given in the text section (https://github.com/revarbat/BOSL2/wiki/shapes2d.scad#module-text); example 6. I've not come across path_text() yet because the library is absolutely massive and I only discovered it recently. It is absolutely amazing; a real game changer, but quite difficult to find specific features. I'll have a look at the path_text() presently. Again, many thanks.

adrianVmariano, 2022-03-30:
I was wondering if there was an example where arc_of was used for
text, but I didn't see it.  I have deleted that obsolete example.