BelfrySCAD/BOSL2 #829: circle/cylinder modules in geometry.scad: make attachable and support all cyl args?

adrianVmariano, 2022-04-03:
The circle modules in geometry.scad don't use attachable and they don't support args like circum.  

Should these modules have the option to make the cylinder at all?

revarbat, 2022-04-04:
How often will these functions NOT be immediately be followed by making a `circle()` or `cylinder()`?

adrianVmariano, 2022-04-04:
Not sure what you mean.  I'm saying that the module could make a circle only and if you want a cylinder, you linear_extrude().  It might simplify the attachment support, though I suppose if you want the children on top or something having it be a full cylinder would be nice.  But I assume you're going to have a problem making them attachable because of their position?  

Should we have a weak form of attachable() that makes colors and diff() etc work but doesn't do any repositioning?  Maybe asserts if there are extra children?

revarbat, 2022-04-04:
Making a circle isn't viable if the inputs are 3D, and the normal UP from the XY plane.

revarbat, 2022-04-04:
I don't see why we can't just use `attachable(anchor,spin.orient, r=r, cp=cp) { circle(); children(); }` for the circle case.
As for the cylinder case, I'm thinking we do something like `rot(from=UP,to=norm) attachable(... r=r, cp=cp) cyl(...)`.

adrianVmariano, 2022-04-04:
Oh, well if attachable takes a cp arg then it seems like it's not so bad, though I do question whether the rot() strategy you outline leaves the user in the dark about how the cylinder is spun.

revarbat, 2022-04-05:
It seems far less confusing than anchor vectors pointing at edges of a cylinder that are NOT aligned vertically (or horizontally).

revarbat, 2022-04-05:
At least you will know how the children will be oriented relative to the cylinder itself.

adrianVmariano, 2022-04-05:
No you won't.  You'll know it up to a mystery rotation of the cylinder around its axis.

adrianVmariano, 2022-04-05:
The alternative would be to define how that rotation will be and use frame_map().

revarbat, 2022-04-05:
I'm unclear what you mean by that.  What I'm envisioning would be just like an attachable cylinder which is auto-oriented and positioned based on the input vectors or points.

revarbat, 2022-04-05:
```openscad

adrianVmariano, 2022-04-05:
The question is what does spin=0 mean and is its meaning obvious/intuitive.  My experience (though I didn't do tests for this case) is that it can be not what you expect when you use rot with from and to.  I very often find that I need to use frame_map instead, because frame map fully controls the mapping and you can define what exactly spin=0 means.

revarbat, 2022-04-05:
I'm stripping geometry creation out of `circle_3points()` and `circle_2tangents()`, and adding `point=` and `corner=` arguments to `circle()` to let you directly do the same for 2D.

For 3D, you can do it yourself with:
```openscad
ci = circle_2tangents(p1, p2, p3, r);
translate(ci[0]) cyl(r=r, h=h, orient=ci[1]);
```
or
```openscad
ci = circle_3points(p1, p2, p3);
translate(ci[0]) cyl(r=ci[1], h=h, orient=ci[2]);
```

revarbat, 2022-04-05:
`arc()` has `points=` to perform the `circle_3points()` equivalent, but does not have `corner=` to do the `circle_2tangents()` equivalent.

revarbat, 2022-04-06:
Implemented in PR #834

closed 2022-04-06 by revarbat