BelfrySCAD/BOSL2 #835: interior_fillet() elongates with

maave0, 2022-04-06:
**Describe the bug**
Using interior_fillet() with angles under 90 causes the shape to elongate. The curved part of the polygon gets shifted away from the origin.

**Expected behavior**
The arc should stay close to the origin with angles under 90. I don't know __exactly__ how others are using this, I'm making a phone case and cutting a shallow curve for curved screens. I tweaked it how I think it should work and called it shallow_fillet()

**Code To Reproduce Bug**
```
include <../BOSL2/std.scad>
include <../BOSL2/shapes3d.scad>

$fn=30;
sampleRadius=4;

// interior_fillet examples, with the curve drifting as angle decreases
interior_fillet(l=1, r=sampleRadius, ang=90);

translate([sampleRadius*1.5,0,0])
interior_fillet(l=1, r=sampleRadius, ang=80);

translate([sampleRadius*3,0,0])
interior_fillet(l=1, r=sampleRadius, ang=70);

translate([sampleRadius*4.5,0,0])
interior_fillet(l=1, r=sampleRadius, ang=60);

translate([sampleRadius*6,0,0])
interior_fillet(l=1, r=sampleRadius, ang=50);

//I don't know how angles greater than 90 should behave
translate([-sampleRadius*1.5,0,0])
interior_fillet(l=1, r=sampleRadius, ang=100);

translate([-sampleRadius*3,0,0])
interior_fillet(l=1, r=sampleRadius, ang=110);

// my version of it
translate([0,sampleRadius*1.5,0])
shallow_fillet(l=1, r=sampleRadius, ang=90);

translate([sampleRadius*1.5,sampleRadius*1.5,0])
shallow_fillet(l=1, r=sampleRadius, ang=80);

translate([sampleRadius*3,sampleRadius*1.5,0])
shallow_fillet(l=1, r=sampleRadius, ang=70);

translate([sampleRadius*4.5,sampleRadius*1.5,0])
shallow_fillet(l=1, r=sampleRadius, ang=60);

translate([sampleRadius*6,sampleRadius*1.5,0])
shallow_fillet(l=1, r=sampleRadius, ang=50);

//I don't know how angles greater than 90 should behave, mine doesn't handle that case either
translate([-sampleRadius*1.5,sampleRadius*1.5,0])
shallow_fillet(l=1, r=sampleRadius, ang=100);

translate([-sampleRadius*3,sampleRadius*1.5,0])
shallow_fillet(l=1, r=sampleRadius, ang=110);

module shallow_fillet(l=1.0, r, ang=90) {
    steps = ceil(segs(r)*ang/360); //based on $fn
    step = ang/steps;
    path = concat(
        [[0,0]],
        [ for (i=[0:1:steps]) 
            let(a=270-i*step) 
            // make a pie slice, then move the arc points up and over
            r*[ cos(a), sin(a) ] + [r*sin(ang), r]
        ]
    );
    linear_extrude(height=l, convexity=10, center=true)
    polygon(path);
}

```
[image]

revarbat, 2022-04-07:
It looks like `interior_fillet()` is intrinsically broken for any angle but 90º.  Luckily I just added the ability to fit an interior corner to `arc()`, which makes this easy to rewrite.

revarbat, 2022-04-07:
Adding example images for non-right angles.

revarbat, 2022-04-07:
Fix implemented.

closed 2022-04-07 by revarbat