BelfrySCAD/BOSL2 #843: non-attachable attachables?

adrianVmariano, 2022-04-12:
In seems like attachable() provides 4 services blended together:  positioning the object, positioning of child objects, tagged operations, and coloring.   Should we have a form of attachable for objects that cannot be repositioned and/or cannot take children----it wouldn't need to know object dimensions----but would still allow tagged operations and colors to work properly?  

Right now there are various non-attachable objects that won't work with tagging or color.

adrianVmariano, 2023-03-03:
```
bottlecaps.scad      bottle_adapter_*()
drawing.scad         stroke(), dashed_stroke()
geometry.scad        hull_points()
mutators.scad        [various]
partitions.scad      partition()
polyhedra.scad       [non-standard]
rounding.scad        convex_offset_extrude(), bent_cutout_mask()
screw_drive.scad     torx_mask2d(), robertson_mask()
screws.scad          screw_head()
shapes2d.scad        text()
shapes3d.scad        text3d()
wiring.scad          wire_bundle()
vnf.scad             vnf_wireframe()
```

revarbat, 2023-03-14:
It's now possible to have a Null/Point attachment geometry where you pass no geometry args (`r=`, `size=`, etc) to `attachable()`/`reorient()`.

adrianVmariano, 2023-03-14:
Yeah, the reason I didn't close this is because of the list above.  I fixed hull_points and text3d but the others not yet.

adrianVmariano, 2023-06-19:
Last ones fixed.

closed 2023-06-19 by adrianVmariano