BelfrySCAD/BOSL2 #863: tube() not working.

sgall17a, 2022-05-02:
Tried and failed with :
    tube(20,7,5);
   // tube(l=30, or=40, ir=35);
   // tube(h=30, ir=35, wall=5);
   // tube(h=30, od=80, id=70);
    tube(20,7,5);
caused assertion error line 1543 bosl2/3dshapes.scad liine
    assert(all_defined([r1, r2, ir1, ir2]), "Must specify two of inner radius/diam, outer radius/diam, and wall width.");

sgall17a, 2022-05-02:
re tube error:  re-downloaded bosl2 today 2/5/22

revarbat, 2022-05-02:
I am unable to reproduce this bug using BOSL2  ver 2.0.652 (~05:45 May 2, 2022 UTC)

adrianVmariano, 2022-05-02:
Also not able to reproduce.  Did you cut and paste the exact code that failed?

sgall17a, 2022-05-02:
include <BOSL2/std.scad>

module oled(){
    panel = [25,15];
    diff(){
    rect_tube(10,panel+[3.2,3.3],panel,anchor=TOP);
    down(1)cube([29,19,1],anchor=TOP);
    back(4)cuboid([12,2,20],$tags="remove");
    }
    
}

//oled();

module switch(){
  //  up(30)tube(20,7,5);
   // tube(l=30, or1=40,or2=40, ir1=35,ir2=35);
//       up(60)tube(h=30, ir=35, wall=5);
   up(90) tube(h=30, od=80, id=70);
}

tube();

DEEPEST. APOLOGIES.

I should have called switch() ---not tube()!  I just couldnt see it at the time.

Stuart Galloway
email: ***@***.***
Phone: 0421560031

adrianVmariano, 2022-05-02:
I get a different error from this:

[WARNING: max() parameter could not be converted in file lib/BOSL2/shapes3d.scad, line 1546](https://github.com/revarbat/BOSL2/issues/1546,/home/adrian/scad/lib/BOSL2/shapes3d.scad) 
[TRACE: called by 'max' in file lib/BOSL2/shapes3d.scad, line 1546](https://github.com/revarbat/BOSL2/issues/1546,/home/adrian/scad/lib/BOSL2/shapes3d.scad) 
[TRACE: called by 'segs' in file lib/BOSL2/shapes3d.scad, line 1546](https://github.com/revarbat/BOSL2/issues/1546,/home/adrian/scad/lib/BOSL2/shapes3d.scad) 
[TRACE: called by 'tube' in file test2.scad, line 24](https://github.com/revarbat/BOSL2/issues/24,/home/adrian/scad/test2.scad)

It may have to do with the defaults being bad?  I'm generally opposed to defaults for dimensions of stuff.

revarbat, 2022-05-02:
Turns out that the assert errors that should be triggered are modules, and thus execute AFTER all the variable setting.  Have to use functional asserts instead.  Thank you OpenSCAD wankery.

revarbat, 2022-05-03:
Made asserts functional in shapes3d.scad and shapes2d.scad.

closed 2022-05-03 by revarbat