BelfrySCAD/BOSL2 #878: [BUG]

LenStruttmann, 2022-05-25:
**Describe the bug**
Documentation error in screws.scad

**Code To Reproduce Bug**
```
Line 61; //   head = head type (see list above).  Default: none
```

**Expected behavior**
I expected to find head type info earlier in the file.  However, that information does not appear to be there.

The head type information seems to be actually defined in the screw_info() module, starting at line 689.

adrianVmariano, 2022-05-26:
Hmmm.  It does say at the top that the parameters are described in the entry for screw_info().  I think screw_info() used to appear first in the file before I reorganized it.  Should all the details in screw_info be moved into screw() and then screw_info() refers to screw()?

adrianVmariano, 2022-05-26:
Another option worth considering: a separate section at the top of the file discussing all the screw parameters.

adrianVmariano, 2022-05-28:
How does the revised screws.scad documentation look?  Any comments?

LenStruttmann, 2022-05-29:
That is much better. I like having all of the specifications at the top.

closed 2022-06-24 by adrianVmariano