BelfrySCAD/BOSL2 #879: Quick query about polygon([points])...

Dorro101, 2022-05-27:
Googled this, but nothing came up. I have become so used to your attachment parameters anchor and spin which have proved indispensable for my parameter based work flow, finding it easier to comprehend and remember descriptive variables than actual measurements. I use the overrides of your replacement modules for 2D primitives like circle() and rect() etc. a fair bit. I was wondering how much work it would be to add these parameters to polygon(). I have also found that shell2d(), which I like to use for adjusting clearance rather than the global $slop variable, seems to only work on individual shapes and not as a group.

If I am being dense or missing the obvious, I would be grateful for a solution . The issue is trivial, so I can happily  live without the arguable convenience of the additional parameters.

Ciao
Dorian

adrianVmariano, 2022-05-27:
Instead of polygon() use region() as a replacement that supports attachment parameters.  

shell2d() works on the geometry that you give it as a unit, so I'm not sure what you mean.

Dorro101, 2022-05-28:
Thanks, having a look at region, should fit the bill. I will have a go. I'm not sure, where or when I encountered this problem, I tried to recreated this quickly, but it worked as I originally expected. I will go through my files to see if I can find the example, when I return home to my desktop. Currently busy away on a quick adventure in Borneo. I am closing this ticket and if I encounter the badly worded issue above, I will file a new ticket...

Many thanks
Dorian

closed 2022-05-28 by Dorro101