BelfrySCAD/BOSL2 #881: torx depths are wrong

adrianVmariano, 2022-05-27:
The torx depths are not meaningful: they are the minimum length of the GO gauge, not a typical recess depth.  Recess depth should be much smaller.

adrianVmariano, 2022-05-30:
Spec on torx, but only on assessing if the recess is correct, not on how to fabricate. 

https://cdn.standards.iteh.ai/samples/63207/f79670cf0a214881aa40b4822580cce3/ISO-10664-2014.pdf

Recess depth in the code appears to be the length of the "GO gauge" and has nothing to do with actual recess depth.  Missing torx values T1-T5, T7, T9, T27 and maybe T35?? should be added, but don't know where to get rounding dimensions.

revarbat, 2022-07-29:
Addressed in PR #923

adrianVmariano, 2022-07-29:
Hex (allen wrench) depth on set screws is half of the flat-to-flat width of the hex.   You still have torx depth bigger than torx diameter.  This causes the torx recess to punch through some screws---it's clearly wrong.  I have not been able to find depth data.

adrianVmariano, 2022-07-29:
[image]

Example CAD from McMaster for a T45 recess in an M10 button head screw.  It's about 2.5 mm deep as shown by ruler.   The diameter is 8mm.   I think you're still falling into the trap of using the GO gauge dimensions and thinking that is the recess dimensions.  That's like thinking the size of the wrench defines the size of the recess.

adrianVmariano, 2022-07-29:
I found that torx socket head screws are defined by ISO 14579.  We could use this data for that table, but it doesn't necessarily apply to any other torx equipped hardware, just socket head screws.   There is a different ISO for cheese head, button head, and countersunk torx "hexalobular" screws.  
[ISO-14579.pdf](https://github.com/revarbat/BOSL2/files/9223129/ISO-14579.pdf)
[](url)

adrianVmariano, 2022-09-03:
This is fixed by latest PR.

closed 2022-09-03 by adrianVmariano