BelfrySCAD/BOSL2 #885: [BUG]Syntax error in geometry.scad

RyantHults, 2022-06-16:
**Describe the bug**
When including BOSL2 into a scad file, I get `ERROR: Parser error in file <path to openscad>/BOSL2/geometry.scad, line 992: syntax error`

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
```

**Expected behavior**
no error

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2019.05 
- Other libraries used: MCAD

RyantHults, 2022-06-16:
Looks like this was a version issue. Updating to 2021.01 fixed it. Closing.

closed 2022-06-16 by RyantHults

WyomingWill, 2022-09-11:
I'm having the same issue, still.... Also, beziers is throwing a syntax error in addition. 

[image]

revarbat, 2022-09-11:
Your version of OpenSCAD is too old and doesn't support the exponent operator. Please upgrade to OpenSCAD 2021.01 or later.

WyomingWill, 2022-09-11:
Thank you! I saw:

`assert(version_num()>=20190500, "BOSL2 requires OpenSCAD version 2019.05 or later.");`

in std.scad and thought I was safe. Works fine with OpenSCAD-2021.01-x86_64.AppImage.

Thanks again for the quick reply.