BelfrySCAD/BOSL2 #893: Slop Calibration Example Fails with "Unknown Module 'Tag'

RAMilewski, 2022-07-09:
I tried to render the Slop Calibration part from the example in _LibFile: constants.scad,_ but it failed with the message:
"Ignoring unknown module "tag"".

I'm fairly new to openSCAD, what am I missing?

adrianVmariano, 2022-07-10:
You need to upgrade to the latest BOSL2.   The tagging code has a massive overhaul recently.

RAMilewski, 2022-07-10:
Downloaded the latest BOSL2 but no joy.   It now fails with:

[WARNING: Too many unnamed arguments supplied in file ../Libraries/BOSL2/attachments.scad, line 850]

Using OpenSCAD 2021.01 Mac OS X

adrianVmariano, 2022-07-10:
I'm mystified.  The slop calibration part works fine for me using the current BOSL2 with Openscad 2021.01.   What is line 850 of attachments.scad?

revarbat, 2022-07-11:
I am unable to reproduce this issue with latest BOSL2 and OpenSCAD 2021.03.08.

RAMilewski, 2022-07-11:
OK... Where do I find 2021.03.08?  All I can see on the openSCAD.org downloads page is 2021.01.

... and on the GitHub openSCAD repository:

[image]

RAMilewski, 2022-07-11:
> I'm mystified. The slop calibration part works fine for me using the current BOSL2 with Openscad 2021.01. What is line 850 of attachments.scad?

Lines 842 through 855 of attachments.scad:

module diff(remove="remove", keep="keep")
{
    req_children($children);
    assert(is_string(remove),"remove must be a string of tags");
    assert(is_string(keep),"keep must be a string of tags"); 
    if (_is_shown())
    {      
        difference() {
            hide(str(remove," ",keep)) children();            <--- Line 850
            show_only(remove) children();
        }
    }
    show_int(keep)children();        
}

module diff(remove="remove", keep="keep")
{
    req_children($children);
    assert(is_string(remove),"remove must be a string of tags");
    assert(is_string(keep),"keep must be a string of tags"); 
    if (_is_shown())
    {      
        difference() {
            hide(str(remove," ",keep)) children();
            show_only(remove) children();
        }
    }
    show_int(keep)children();        
}

adrianVmariano, 2022-07-11:
Note that it should work with 2021.01, which is the stable version.  The new version is a development release.

RAMilewski, 2022-07-12:
OK.... MY BAD!   It was a copy & paste error from the example into my SCAD workspace!
Apologies for the hassle!

closed 2022-07-12 by RAMilewski