BelfrySCAD/BOSL2 #894: [BUG]  Cuboid: chamfer single edge fails unpredictably

IanM-2022, 2022-07-10:
[report chamfer error.zip](https://github.com/revarbat/BOSL2/files/9079797/report.chamfer.error.zip)
### [image]
**Expected:**      thin square cuboid with one vertical edge chamfered.

**Actual result:** at various unpredictable sizes it toggles between the expected result and a thin isosceles right triangular prism as chamfer removes half the square, on the diagonal between the two adjacent vertical edges to the intended one!
e.g. it fails at D=20.1 or 19.7, but not at D=19.8 to 20.0

```
include <BOSL2/std.scad>
// change D slightly to see randomness!
D=20.1;//[10:0.1:30] 
cuboid([D,D,D/10], chamfer=D/20, edges=[FRONT+LEFT]);
```
- OpenSCAD Version:  2.0.652 downloaded 2022-07-10
- Other libraries used: none
- 
p.s. *WHY* isn't BOSL_VERSION getting 'bumped'? Its been at 2.0.652 since 2021!

adrianVmariano, 2022-07-10:
I duplicated this bug.  I also found that the bug does NOT occur with roundings, only with chamfers.  

Most cases are handled wrong.  The correctly handled cases are not at all random.  They are 10, 10.7,10.8, 11.5,11.6,12.4,12.5, 13.2, 13.3, 14, 14.1, 14.9, 15, 15.7,15.8, 16.5,16.6, 17.4,17.5, 18.2,18.3, 19,19.1, 19.9, 20, 20.7, 20.8, ...

revarbat, 2022-07-11:
Fixed.

closed 2022-07-11 by revarbat