BelfrySCAD/BOSL2 #900: Add a constant for typographers points.

RAMilewski, 2022-07-15:
**Is your feature request related to a problem? Please describe.**
The text3d.scad is a bit confusing in that the size parameter is referred to as "font size".   It actually means the size of the type in whatever the base unit of OpenScad is (normally mm).  In the world of computer type,  "font size" is traditionally measures in "points"  usually 1/72 inch or about 0.353 mm.  

**Describe the solution you'd like**
Add a constant PT or PTS to constands.scad

**Describe alternatives you've considered**
Alternatively adding a note to the documentation pointing out that the size parameter means mm.

**Example Code**
PT = INCH/72;
or
PT = 0.35278;

**Additional context**
[Wikipedia article on Typographic Points](https://en.wikipedia.org/wiki/Point_(typography))

adrianVmariano, 2022-07-16:
My initial reaction to this request is that it's not really very useful and PT is a generic variable standing for "point" that also has a geometric interpretation.   Why do I think it's not useful?  Because OpenSCAD isn't a typesetting system: specifying font size in points doesn't really make sense if I'm trying to figure out what font is going to allow my text to fit on an object that has been designed in millimeters.  

If you think that you can make a font with size 10*PT and get a "10 point font" you're actually wrong.  There's no way to get a 10 point font in OpenSCAD.  See here for what font point size means in typography:  https://www.onlineprinters.co.uk/magazine/font-sizes/  Note that it's the size of a bounding box for the font.  In OpenSCAD font size is the size of the tallest ascender above the baseline, so it excludes the margins of the bounding box and it excludes the descender size.  (There may be an iterative way to get an actual font point size using the dev version textmetrics code, but that's sort of complicated.)  

It seems like some additional documentation of what the "size" parameter means would be reasonable, though.  (Note that the "size" parameter in text3d is just based to the base OpenSCAD text() module.)  

Final decision on whether to add PT or something like it will be made by Revar.

revarbat, 2022-07-16:
I think providing a PT constant would be disingenuous, as `text(size=12*PT...)` (and thus `text3d()`) wouldn't actually provide text in a 12pt font, and the PT constant would have to be different based on which font you are using, to even approximate that.

RAMilewski, 2022-07-20:
OK.  I understand the confusion that would lead to.  In reality specifying a 12pt font doesn't yield the same size characters across different fonts in desktop publishing either.  But... I do think a line in the docs for text() and text3d() pointing out that size is specified in OpenSCAD base units would be helpful.

closed 2022-08-07 by RAMilewski