BelfrySCAD/BOSL2 #930: Example 3 in chamfer_edge_mask apparently not working. [BUG]

RAMilewski, 2022-08-21:
**Describe the bug**
When I remove the # from the example code, I get this: 
[image]

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>

diff("mask")
cube(50, center=true) {
    edge_mask(TOP+RIGHT)
        chamfer_edge_mask(l=50, chamfer=10);
}
	
```
**Expected behavior**
It should chamfer the edge. 

- OpenSCAD Version:   2021.01
- Other libraries used:  None.

revarbat, 2022-08-21:
The mask tags changed a couple months back. Instead of `diff("mask")`, use `diff()` with no argument.

revarbat, 2022-08-21:
The examples code *is* wrong, though.

revarbat, 2022-08-21:
Fixes committed to PR #929

closed 2022-08-21 by revarbat