BelfrySCAD/BOSL2 #937: WARNING: variable edges not specified as parameter

RAMilewski, 2022-08-31:
I'm not sure whether I should worry about this warning or not.   I can't find any other references to "variable edges".

Here's a code code snippet that throws the warning:

`include <BOSL2/std.scad>

gnss = [25,25,6.5];
pad  = [4.4, gnss.y * 0.6, gnss.y * 0.6];
pad2  = [4.4, gnss.y * 0.3, gnss.y * 0.4];

prismoid(size1 = [pad.y,pad.z], size2 = [pad2.y, pad2.z], shift = [(pad2.y - pad.y)/2,0],
			h = pad.x, rounding = 2, edges = "Z", anchor = LEFT+BOT);
`

adrianVmariano, 2022-09-03:
What you are doing is passing the "edges" parameter to prismoid(), which doesn't accept that parameter.   OpenSCAD alerts you that edges is not a parameter with that warning.   Note that prismoid can **only** round the "Z" edges, so no edges parameter is used.

RAMilewski, 2022-09-08:
Aha!  Noob pilot error.  Thank you.

closed 2022-09-08 by RAMilewski