BelfrySCAD/BOSL2 #941: [BUG] ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.

rola25, 2022-09-09:
**Describe the bug**
When I try to render the example code down below, I get ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.

**Code To Reproduce Bug**
```
// Code goes here.

include <BOSL2/std.scad>

$fn = 50;
fn = 50;

tube_diameter = 16;
tubespin = 110;
width_holder = 20;
angle = 235;
endcaps="round"; // [round, butt, chisel, square, dot, block, diamond]
holder_thick = 4;
basement_thick = holder_thick;
basement_round=3.2;
basement_len=180;

/* [Hidden] */
basementX = basement_len + holder_thick;

module arctube(l=10,r,d,width=5,angle=180, endcaps="butt", anchor=CENTER, spin=0, orient=UP) {
r = get_radius(r=r,d=d,dflt=1) + width/2;
attachable(anchor,spin,orient, r=r, l=l) {
path = arc(r=r, angle=angle);
linear_extrude(height=l, center=true, convexity=10)
stroke(path, width=width, endcaps=endcaps);
children();
}
}

union(){
myspin = tubespin + ((360-angle)/2);
cuboid([basementX,basement_thick,width_holder], rounding=basement_round, edges="Z",except_edges=[BACK], anchor=BACK+LEFT)
position(LEFT+TOP)
back(tube_diameter/2+holder_thick/2)
arctube(l=width_holder,d=tube_diameter,width=holder_thick, angle=angle, endcaps=endcaps, spin=myspin, anchor=TOP);
}

adrianVmariano, 2022-09-09:
That is a surprising problem.  If you want a quick work around, include rounding.scad and use `offset_stroke(path, width=width, start="round", end="round");` to replace your stroke call.

adrianVmariano, 2022-09-12:
Here's a minimal example:

    include <BOSL2/std.scad>

    $fn = 10;      // Under 10 and problem goes away
    r=10;          // Goes away if r=11 or higher
    angle = 180;   // Under 180 and problem goes away
    width = 1.5;   // Problem goes away if width is 1

    cuboid(13);
    path = (arc(r=r, angle=angle));
    linear_extrude(height=1)
      stroke(path, width=width,$fn=12);

I believe this is fixed in PR 944, but @revarbat should have a look to see that I didn't break something.  Problem is that the circular wedges that join the segments are exact, which seems to lead to zero width slots in the polygon before extrusion.  I thought that wasn't supposed to be a problem in 2d, but I guess it is.

adrianVmariano, 2022-09-14:
This is now fixed by the latest merge.

closed 2022-09-14 by adrianVmariano