BelfrySCAD/BOSL2 #942: Acme_threaded_nut() argument list incomplete

RAMilewski, 2022-09-10:
The starts argument is missing from the "By name" list of arguments for acme_threaded_nut() in the wiki.

It appears in some of the examples, but not in the argument list.

ps: is this the correct place to report issues with the wiki?

adrianVmariano, 2022-09-10:
Yes, this is the right place to report problems with the wiki.   I fixed this in my fork.  I also checked to see if this error appeared elsewhere, and I noticed that buttress_threaded_nut and rod also don't list starts as an option---but they don't accept it.  Is there some reason that the buttress thread shouldn't take the starts argument?  @revarbat ?

adrianVmariano, 2022-09-11:
This is fixed in PR 943.   I also added starts to buttress threads.

closed 2022-09-12 by RAMilewski