BelfrySCAD/BOSL2 #955: cyl() needs internal diam/radius args

adrianVmariano, 2022-09-22:
There should be id/ir args for cyl that specify internal diameter or radius.

revarbat, 2022-09-23:
So that `cyl(l=10,ir=10);` is the same as `cyl(l=10,r=10,circum=true);`?

adrianVmariano, 2022-09-23:
I didn't see the circum arg.  (And I was even looking for it.)    I guess it's fine, then.  I don't know if there's ever a case where someone would want `id1=1, od2=20` which would give a bit more flexibility.

adrianVmariano, 2022-09-23:
I guess there is a question about API consistency here, though, since some modules use `id` and some use `circum`.

adrianVmariano, 2022-09-23:
I suppose one solution to this problem might be to make all the modules support both forms.

adrianVmariano, 2022-11-06:
I took a look at modules and current interface.  The only one that supports id/od is regular_ngon.  Everything else just takes circum.

closed 2022-11-06 by adrianVmariano

revarbat, 2022-11-06:
Which begs the question, should `regular_ngon()` use `circum=`?

adrianVmariano, 2022-11-06:
Well, regular n_gon is making a regular n_gon.  It makes it the exact size you say.  So the concept of circumscribing doesn't really make sense.  The other operations are all making rounded forms that are being approximated, so saying that the approximation circumscribes the true form makes sense.

Other modules with id/ir include star and tube, where it also makes sense that there is an inner measurement.

adrianVmariano, 2022-11-06:
We could add circum as an option to regular_ngon just for consistency, and leave ir/id available also.