BelfrySCAD/BOSL2 #956: Add mitre and convex fillet to interior_fillet

roel-v, 2022-09-23:
From what I can tell, interior_fillet() right now only does concave fillets. I think it would be useful (it would be to me :) ) to add mitre- and convex fillet variations.

revarbat, 2022-09-24:
An interior chamfer is just a cube rotated 45 degrees and perhaps scaled by sqrt(2).
More complex fillet shapes are easily generated by `linear_extrude()`ing the shapes in masks2d.scad.  That'll give you coves, chamfers, roundovers, and even arbitrarily complex ogees.

adrianVmariano, 2022-09-24:
Are there examples of doing this?

revarbat, 2022-09-25:
Fair enough.  Adding examples and docs.

roel-v, 2022-09-25:
I understand that the mitre fillet is just that (it's what I usually do now), although it does get more complicated with large fillets on thin pieces, where the cube 'sticks out' at the other side of the corner that is filleted. But even then, it's still manageable. And yes also other fillets can be made with various permutations of triangles, circle slices or triangles minus circle slices and linear_extrude() - It's just that I see BOSL (and I think that's how it's meant to be) as a library of higher-level geometric building blocks, to be used in place of geometrical operations, so as to make (somewhat) standardized and self-documenting designs. And since there is an interior_fillet(), it seemed (to me) that the ontologically correct situation would be to have interior_fillet_concave(), interior_fillet_mitre() and interior_fillet_convex(), all with the same API and conceptual integration into the rest of BOSL.

But it was just a suggestion. Thanks for BOSL, I like it a lot.

PR #958 (2022-09-25): Documented usage of 2D masks to make interior angle decorations.
Fixes #956 documentation issue.
closed 2022-09-25 by revarbat