BelfrySCAD/BOSL2 #965: [BUG]

LenStruttmann, 2022-10-08:
**Describe the bug**
Using screws.scad, it appears that one cannot make metric nuts.  Error received:

[WARNING: Ignoring unknown function '_nut_info_metric' in file screws.scad, line 1720](https://github.com/revarbat/BOSL2/issues/1720,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[ERROR: Assertion 'is_list(list)' failed: "Invalid list in list_remove" in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/lists.scad, line 712](https://github.com/revarbat/BOSL2/issues/712,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/lists.scad) 
[TRACE: called by 'list_remove' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/structs.scad, line 52](https://github.com/revarbat/BOSL2/issues/52,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/structs.scad) 
[TRACE: called by 'concat' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/structs.scad, line 52](https://github.com/revarbat/BOSL2/issues/52,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/structs.scad) 
[TRACE: called by 'struct_set' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad, line 469](https://github.com/revarbat/BOSL2/issues/469,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[TRACE: called by '_struct_reset' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad, line 1723](https://github.com/revarbat/BOSL2/issues/1723,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[TRACE: called by 'nut_info' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad, line 439](https://github.com/revarbat/BOSL2/issues/439,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[TRACE: called by '_get_spec' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad, line 1407](https://github.com/revarbat/BOSL2/issues/1407,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[TRACE: called by 'nut' in file paintCaddy2.scad, line 15](https://github.com/revarbat/BOSL2/issues/15,D:/OpenSCAD/paintCaddy/paintCaddy2.scad) 

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/screws.scad>

nut( "M8" );  //  This is the example in line 1384 of screws.scad

```

**Expected behavior**
A metric nut to be created.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version:
- Other libraries used:

adrianVmariano, 2022-10-10:
To add a little more, the DIN standards agree with ISO about nut width---except for 4 sizes.  But they disagree quite a bit on nut thickness.  And while DIN has just one standard for regular nuts that I'm aware of, it has two different standards for thin nuts that conflict with each other.   ISO doesn't seem to specify square nuts.  DIN does, but for two sizes gives two different options on the nut width.

LenStruttmann, 2022-10-10:
Hi! Since nut standards are so chaotic, and since you use ISO to create screws, then I suggest just using the ISO standard and document that DIN standards are conflicting.  Your previous two messages are a good start at telling people why you use ISO. The module nut() allows DIN users to specify all of the physical parameters.

adrianVmariano, 2022-10-10:
I'd be a little happier with that solution if I actually knew a place I could buy ISO hardware.   That is, if DIN is actually dominant then following ISO may be a mistake, even if justified based on how things are currently organized.  (Do ISO nuts actually exist?)  Also doing strict ISO would mean completely excluding square nuts from the metric support, since they aren't in ISO at all.  For screws it appears that DIN and ISO are more or less in agreement: on McMaster they list the hardware as simultaneously complying with both.  So this problem didn't arise there.  (And you could argue that screws are also DIN.)

LenStruttmann, 2022-10-10:
Hmmm... I agree that not having square nuts would be an inconvenience. I think that it is interesting that, after falling out of fashion for years, with the advent of 3D printed but traps, square nuts are in higher demand now. I wonder what that source and spec are of the square nuts that Prusa uses in their printers.

LenStruttmann, 2022-10-10:
So, after some thought, I guess that it all depends on who are the dominant users of BOSL2.  If it's primarily meant to be used by hobbyists designing for 3D printers, I would vote for full DIN.  These users (of which I am one) don't need high precision due to the current limitations of 3D printers.  Even threading is superfluous for purchased fasteners.  

However, if BOLS2's main user base consists of professional engineers and designers who require high precision, then you may want to include BOTH ISO and DIN, which would require a less simple interface for the user.

Just some thoughts.

adrianVmariano, 2022-10-10:
The nuts in my prusa appear to comply with the DIN square nut standard with the thinner nut thickness.     The Prusa is a fairly accurate machine and the discrepancies between DIN and ISO can be millimeters, which is far beyond the errors in the printer, which are at the 0.05 mm level.  This makes it seem like some scheme for including both is necessary.  I was thinking of a thickness="undersize" option to get the DIN thicknesses, but that's more confusing if there are multiple options.  I thought about thickness="normal din" or "din normal", or "din thin" etc.  This doesn't address DIN have conflicting thin nut standards, but I could ignore the one that looks less popular.  However, this more seriously doesn't deal with the four nuts whose width is also different in DIN.   

Note that I have used threading in several models that I printed with the prusa, though at a larger scale.

adrianVmariano, 2022-10-10:
Another option would be to rename "ISO" to something else like "metric" and just use only the DIN standard for nuts.

LenStruttmann, 2022-10-10:
"the discrepancies between DIN and ISO can be millimeters"    Wow!  I presume that large of a discrepancy is for the larger nuts, not for, say, M2 - M5.

"Another option would be to rename "ISO" to something else like "metric" and just use only the DIN standard for nuts."
That works for me.  All long as it is well documented, there should not be any problems with that.

Oh yes, threading is needed, just not normally when working with small purchased fasteners.

adrianVmariano, 2022-10-10:
For M5 the max ISO thickness is 4.7 and from DIN 4.0, though it turns out that the M5 discrepancy is anomalously large.   For M6 ISO is 5.2 and DIN is 5.  For thin nuts...Hmmm.  I hadn't realized but it looks like the thin ISO agrees with one of the thin DIN values.   But DIN's internal disagreement gives a thickness of either 5mm or 6mm for M10, depending on if you use DIN 439 or DIN936.  

At really big nuts we have M24 with ISO thickness 21.5 and DIN thickness 19.  

For M10 ISO specifies 16mm width and DIN 17mm width.

LenStruttmann, 2022-10-11:
Yes, I can see why you are stuck.  No matter what you end up doing, it will be a compromise.  I guess that there isn't enough incentive for manufacturers to switch to ISO, even with all of the benefits of a common consistent standard,  For me, it's not such a big deal to design my own nuts and nut traps based upon the measurement for the parts that I have.

If you want to close this issue without deciding, just update the documentation in screws.scad to reflect that metric nuts are not yet available.  I'll be glad to open a Feature Request for the addition of metric nuts just to keep people from opening more BUG and Feature Request issues.  

Again, thanks for all of the work you've done on BOSL2.  It is appreciated.

adrianVmariano, 2022-10-11:
I would like to reach a conclusion on this and be able to finish it off somehow, so there's no reason to close the issue until that happens.  Note that ASME also defines metric nuts and it looks like they follow ISO, not DIN.

adrianVmariano, 2022-10-13:
This is fixed in the current version.

closed 2022-10-13 by adrianVmariano