BelfrySCAD/BOSL2 #966: screws.scad: Request that echo_struct() calls be optional

LenStruttmann, 2022-10-09:
**Is your feature request related to a problem? Please describe.**
When creating a model with many screws, nuts, nut_traps, etc, the OpenSCAD log quickly gets filled with screw_info and nut-info structs echoed by echo_struct().  This noise in the log makes debugging more difficult.

**Describe the solution you'd like**
Perhaps a parameter passed to screw(), nut(), etc., that allows the user to control whether or not echo_struct() gets called. 

**Describe alternatives you've considered**
I've gone into the source and commented out the echo_struct() calls, but that's not the best solution.

**Example Code**
```
nut_trap_side(15, "#8", $slop=0.1);
```

**Additional context**
Add any other context or screenshots about the feature request here.

adrianVmariano, 2022-10-10:
The echo struct calls are going to be gone completely sometime soon, I think.  They are there because screws.scad was not really in a final state and needed work, development and debugging.  Also note I see you're using the new nut trap code, so if you have any comments on how that works, I'd be interested.

LenStruttmann, 2022-10-10:
That's good about echo_struct(), although I wouldn't mind having an argument, such as debug=true, to see the structs.  With that in mind, feel free to close this issue.

The nut traps seem to work well, I'm just now starting to play with them.  I was able to approximate an M2.5 nut trap by using "#3" with a slop of 0.3. 

Also, I greatly appreciate that I can use the same screw spec for screw(), nut(), screw_hole(), and nut_trap() and with the default orientation and anchor they are all aligned perfectly.

For example:
include <BOSL2/std.scad>
include <BOSL2/screws.scad>

screwSpec = "#8-32,1/2";
screwLength = 10;

nutTrapSlop = 0.1;
nutTrapLength = 10;

screw( screwSpec, head="flat", thread=0, length=screwLength );
#screw_hole( screwSpec, head="flat", thread=0, length=screwLength );
nut(screwSpec, $slop=nutTrapSlop);
#nut_trap_side(nutTrapLength, screwSpec, $slop=nutTrapSlop);

Lastly, the only other thing that I could wish for would be the ability to create teardrop shaped screw holes for horizontal screw holes to overcome the drooping that occurs when using FDM printers.  Should I put that in as a low priority feature request?

Thanks for all the hard work.  BOSL2 is amazing (if sometimes overwhelming)  :-)

adrianVmariano, 2022-10-13:
fixed in the current version.

closed 2022-10-13 by adrianVmariano