BelfrySCAD/BOSL2 #980: Faces are missing in the result of vnf_halfspace

herdima, 2022-10-28:
In all three generated intersections, faces are missing in the mid of the intersection area:
`vnf=sphere(r=50, style="icosa", $fn=16);
%vnf_polyhedron(vnf);
hsps= [[0,1,0,0],[0,0,-1,0],[-1,0,0,0]];
for(i=idx(hsps))
translate([110*(i+1),0,0])
vnf_polyhedron(vnf_halfspace(hsps[i], vnf));`
I did not find smaller examples for this behavior when experimenting with tetrahedra, cubes and various double cones.
[image]

adrianVmariano, 2022-11-01:
Seems to be a rounding problem.  Looks like solution is to change

         inside = [for(x=vnf[0]) plane*[each x,-1] >= 0 ? 1 : 0],

to

         inside = [for(x=vnf[0]) plane*[each x,-1] >= -EPSILON ? 1 : 0],

in vnf_halfspace().

adrianVmariano, 2022-11-03:
Fixed in #983

closed 2022-11-05 by adrianVmariano