BelfrySCAD/BOSL2 #981: [bselect can fail in its for-clause]

herdima, 2022-10-28:
**Describe the bug**
When called with an empty list then the function bselect of lists.scad causes a DEPRECATED message due to a for clause of for(i=[0:-1]). Respective calls can occur internally in BOSL2.

**Buggy code**
```
function bselect(list,index) =
    assert(is_list(list)||is_string(list), "Improper list." )
    assert(is_list(index) && len(index)>=len(list) , "Improper index list." )
    is_string(list)? str_join(bselect( [for (x=list) x], index)) :
    //[for(i=[0:len(list)-1]) if (index[i]) list[i]];
    [for(i=[0:len(list)-1]) if (index[i]) list[i]];
```

**Fixed code**
```
function bselect(list,index) =
    assert(is_list(list)||is_string(list), "Improper list." )
    assert(is_list(index) && len(index)>=len(list) , "Improper index list." )
    is_string(list)? str_join(bselect( [for (x=list) x], index)) :
    //[for(i=[0:len(list)-1]) if (index[i]) list[i]];
    [for(i=[0:1:len(list)-1]) if (index[i]) list[i]];
```

**Additional context**
- OpenSCAD Version: 2021.01
- Other libraries used:
- System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 12 CPUs 31.87 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit

adrianVmariano, 2022-11-01:
That looks like the right fix.  In looking at the code, I also wonder if there's a reason to allow the second parameter (which seems badly named) to have a length longer than the input list.  That seems questionable to me.

adrianVmariano, 2022-11-03:
Fixed in #983.

closed 2022-11-05 by adrianVmariano