BelfrySCAD/BOSL2 #982: vnf_triangulate fails on degenerate quad

herdima, 2022-11-01:
**Describe the bug**
vnf_triangulate does not behave as expected for a face with four or more vertices when the vertices of the face are collinear.
Such a face can arise e.g. when applying vnf_halfspace to a result of vnf_region.

**Code To Reproduce Bug**
```
vnf = [[[0,0,0], [6,0,0], [9,0,0], [3,0,0]], [[0,1,2,3]]];
// Works and produces appropriate warnings:
vnf_validate(vnf);    
//vnf_wireframe(vnf); // Works
//debug_vnf(vnf);     // Fails
// Produces a cryptic warning from geometry.scad
// and subsequently fails in an assertion:
vnf2 = vnf_triangulate(vnf);
```
**Expected behavior**
For such cases, vnf_triangulate should not produce any warning and it should return an empty VNF.

adrianVmariano, 2022-11-02:
I think actually the VNF is invalid, so it's not clear that vnf_triangulate should work: vnf_validate reports errors.  

What is the example where vnf_halfspace() produces a face like this?   Note that vnf_halfspace is meant to be run on a closed vnf that represents a polyhedron.  If you feed it the output of vnf_region that's not a closed surface.  I expect the result will be bad unless you set closed=false in the call to vnf_halfspace().  

I suspect that this is a case where trying to verify, e.g., that the vnf is a closed object might be time consuming, and hence not something we want to do in general in vnf_halfspace().   I wonder if we should have a function that checks this.   Maybe docs need to be improved?  

Report back on whether your problem is resolved by using vnf_halfspace() with closed=false, or whether there's a more serious problem.

herdima, 2022-11-02:
vnf_halfspace() with closed=false resolves the problem. From the docs is was not clear to me that it would.

```// Example taken from vnf.scad, vnf_from_region:
  region = [square([20,10],center=true),
    right(5,square(4,center=true)),
    left(5,square(6,center=true))];
  vnf = vnf_from_region(region);
  color("gray")down(.125)
  #linear_extrude(height=.125)region(region);
  //vnf_wireframe(vnf,width=.25);

  // Intersection with halfspace:
  hs = [1,0,0,5];
  cube = cube(30*[1,1,1], center=true);
  //#vnf_polyhedron(cube);
  //%vnf_polyhedron(vnf_halfspace(hs, cube));
  //vnf2 = vnf_halfspace(hs, vnf, closed=true); // Fails
  vnf2 = vnf_halfspace(hs, vnf, closed=false);  // Works
  up(1) vnf_polyhedron(vnf2);
```
[image]

adrianVmariano, 2022-11-02:
Ok.  I will improve the docs.  

Note that for your example you'd be better off using difference() or intersection() to operate on the polygon first and then use vnf_from_region().

adrianVmariano, 2022-11-03:
This is fixed in #983

closed 2022-11-05 by adrianVmariano