BelfrySCAD/BOSL2 #987: vnf_merge does not work correctly for eps=0

herdima, 2022-11-07:
**Describe the bug**
vnf_merge produces a warning but no VNF when called with eps=0. This is because the case r=0 is not correctly handled in the function vector_search. Some expressions there use "...<r" where they should use "...<=r". 

**Code To Reproduce Bug**
```
  P = [[0,0,3], [9,0,0], [0,9,6], [-9,0,0]];
  vnf1 = [P, [[0,2,1]]];
  vnf2 = [P, [[0,3,2]]];
  vnf3 = vnf_join([vnf1, vnf2]);
  vnf4 = vnf_merge_points(vnf3, eps=0);
```

**Expected behavior**
Behavior and result should be the same as for vnf_merge_points(vnf3).

adrianVmariano, 2022-11-07:
Why do you need eps equal to zero?  It will fail except on specially crafted toy cases, I think.

adrianVmariano, 2022-11-08:
Ok.  Hopefully this if fixed in latest PR---issue is with vector_search on small sets.

closed 2022-11-10 by adrianVmariano

adrianVmariano, 2022-11-10:
I said it was fixed in the latest PR.  If you find otherwise, reopen the issue.  

It's related to small sets because it appears only the `len(target)<=400` case has the problem.