BelfrySCAD/BOSL2 #1004: [BUG] Small changes in h/w1/w2 break anchoring on rounded trapezoid.

FarmerDave, 2022-12-12:
While tweaking the size of a trapezoid with anchor=BACK+LEFT, I noticed that the location "jumped" as I tweaked the values of h, w1, and w2.  Instead of anchoring BACK+LEFT, it behaves as if anchor is set to LEFT.  Doesn't occur if not rounding.  Also occurs if per-corner rounding is used.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>

trapezoid(h=12.6, w1=61.1, w2=36, rounding=1, anchor=BACK+LEFT);
#trapezoid(h=12.6, w1=61.5, w2=36, rounding=1, anchor=BACK+LEFT);
```
**Expected behavior**
The two trapezoids should appear in more or less the same location.

**Screenshots**
[image]

**Additional context**
- OpenSCAD Version: 2021.01 (Windows)
- Other libraries used: none

adrianVmariano, 2022-12-14:
This is happening because trapezoid uses path style anchoring, which is sensitive to the angle of the trapezoid sides.   It looks like this type of anchoring is used only when you turn on rounding.  

This same issue arose with rect().  I think we need trapezoid to also accept "box" and "perim" atypes and with "perim" it seems like the anchors should still be actually on the corners, not the result of the path anchor.

adrianVmariano, 2022-12-14:
Do we have something in a tutorial or somewhere that explains how path anchoring works, because it's sort of non-obvious in cases like this.

adrianVmariano, 2022-12-21:
I think that attachments to roundings on rect() are also broken, though in fewer situations.  But look at these anchors:

[image]

I would say that's pretty wacky, with the side anchors not being centered and in one case the side anchor at the corner.  I think that we need to add a mechanism for specifying adjusted corner positions and then pass in a list of corner points that supersede the basic ones for both trapezoid and rect.  

Another question about trapezoid?  The angle option is very non-intuitive...arguably just plain wrong.  Why would you want to set the angle against the Y axis instead of setting the actual angle of the trapezoid you're making.    I feel like this needs to be fixed, but it is a breaking change.   I guess we could add the correct option and call it something else?

adrianVmariano, 2023-02-12:
Note that the original problem should be fixed---trapezoid anchoring is now with respect to the unrounded object.  Some reasonable handling of rounded trapezoids (and rects) awaits implementation of a function/table method for giving the anchors of attachable objects.

closed 2023-03-11 by revarbat