BelfrySCAD/BOSL2 #1007: Doing `cuboid([...], rounding=..., teardrop=90)` causes error, when it should actually disable teardrop

DeflateAwning, 2022-12-16:
**Describe the bug**
In cuboid (and maybe other functions), setting `teardrop=90` causes an error, when it should actually behave the same as `teardrop=false`.

**Code To Reproduce Bug**
```
front(30) cuboid([10, 10, 10], rounding=2, teardrop=89.999); // this works
front(-30) cuboid([10, 10, 10], rounding=2, teardrop=90); // this errors, when it should actually just work the same as teardrop=false
```

adrianVmariano, 2022-12-16:
Why should teardrop=90 behave the same as teardrop=false?   This would require special code to check this case and give it special handling, with no obvious benefit.

revarbat, 2022-12-16:
My instinct is that it should assert an error on a value of 90, as that is an obviously invalid value.

DeflateAwning, 2022-12-16:
Shouldn't 90 degrees do approximately the same thing as 89.99999? It seems pretty clear to me that 90 degrees means to disable this functionality, as the trend is that as numbers get closer to 90, the effect is less, until the minimum effect (nil) at 90 degrees.

adrianVmariano, 2023-01-03:
We should close this issue, right?  We're not going to make teardrop valid at its limiting values.

revarbat, 2023-01-06:
Eh.  Turns out if you just make the assert teardrop<=90 (instead of <90) then it just works.  This is a harmless request that I see no problems with granting.

closed 2023-01-06 by revarbat