BelfrySCAD/BOSL2 #1008: [BUG] regression in rect_tube with shift inbetween 2205 and 2212

bablokb, 2022-12-16:
I recently updated my BOSL2 version. Former version was downloaded May 06.2022. New version Dec 16.2022.

I'm using rect_tube with the shift parameter like this:
```
x1_panel = 85.6; 
y1_panel = 48.7;
r_panel = 3; 
z_body = 30;
w4 = 1.67;

y_corpus_bot   = y1_panel/3;
y_corpus_top   = y_corpus_bot*2/3;
y_corpus_shift = (y_corpus_bot-y_corpus_top)/2; 
h_corpus       = z_body + 20;

module corpus() {
  rect_tube(size1=[x1_panel,y_corpus_bot],size2=[x1_panel,y_corpus_top],
                   h=h_corpus,wall=w4,shift=[0,-y_corpus_shift],
                   rounding=r_panel,irounding=r_panel,anchor=BOTTOM+CENTER);
}
```
This is a rect_tube with a smaller top shifted to the front so that I have one orthogonal wall. The first image shows the result with the older version from May, the second the result with the current version. IMHO the old version works correct.

Note that there must be additional issues since in the current version also the connectors (on the left) are not connected to the corpus.

[image]
[image]

**Additional context**
- OpenSCAD Version: 2021.01
- complete version of the code (file `3D/badger2040-support.scad`): https://github.com/bablokb/circuitpython-clock

revarbat, 2022-12-17:
Much simplified example of shift error:
```openscad
  rect_tube(
    size1 = [16, 16],
    size2 = [16, 10],
    h = 20, wall = 2,
    shift = [0, -3],
    anchor=BOTTOM+CENTER
  );
```

revarbat, 2022-12-17:
`rect_tube()` is internally using attachments so it can use `diff()` to remove the inner `prismoid()` from the outer `prismoid()`. This is slightly problematic for using `shift=` as that offsets the CENTER anchor position such that the way the `prismoids()` were attached gets half an XY shift translation.  This can be fixed by re-attaching based on the `BOTTOM`, and going `down(h/2)`, but instead I'm going to change to using `difference()`, as there's a weirdness in the current code where `children()` are apparently attached twice.

closed 2022-12-17 by revarbat