BelfrySCAD/BOSL2 #1013: [BUG] Rendering snap pins only shows half the structure in OpenSCAD UI

dclaze, 2022-12-26:
**Describe the bug**
Rendering snap pins only renders half the structure when using OpenScad UI.

> NOTE: Rendering appears to work as expected using CLI command.

1. Open OpenSCAD in UI
2. Paste below code
3. Click render button
4. See only half of snap pin

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>;
include <BOSL2/joiners.scad>;

snap_pin("small");
```

**Expected behavior**
Expected to render double sided pin, but only half is rendered

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2022.03.02
- Other libraries used: N/A

adrianVmariano, 2023-01-02:
I am not able to duplicate your problem.  Your code produces the full pin both on preview or render for me, either in the stable 2019 version or in the nightly build.  (I don't have 2022.03.02).  

Side note:  your $fn is too small.

revarbat, 2023-01-03:
I also cannot reproduce it in OpenSCAD 2022.08.22 (git ea8950bd3).
On an unrelated note, the pointy ends have too much overhang to print cleanly without support.  I don't see any reason that they *need* to be that pointy, though.  If the points were pulled back to be at a 45º overhang, they should print without support rather nicely.

revarbat, 2023-01-03:
This fixes the overhang, I think:
```diff
diff --git a/joiners.scad b/joiners.scad
index 38bdfa0..7cf7a98 100644
--- a/joiners.scad
+++ b/joiners.scad
@@ -754,8 +754,10 @@ module _pin_shaft(r, lStraight, nub, nubscale, stretch, d, pointed)
    down(extra) cylinder(r = r, h = lStraight + extra);
    up(lStraight) {
       zscale(stretch) {
-         sphere(r = r);
-         if (pointed) up(rPoint) cylinder(r1 = rPoint, r2 = 0, h = rPoint);
+         hull() {
+            sphere(r = r);
+            if (pointed) up(rPoint) cylinder(r1 = rPoint, r2 = 0, h = rPoint/stretch);
+         }
       }
    }
    up(d) yscale(nubscale) _pin_nub(r = r, nub = nub, h = lStraight - d);
```

adrianVmariano, 2023-01-03:
I have printed it as it is and don't recall any problems.  It was based on a design that had been tested by multiple people, so I didn't change the design.  But I guess if you're certain it's an improvement and the pins will still fit well in the shaft....

Actually, your change seems to make almost no difference when a proper $fn value is used:

Original form:

[image]

With added hull():

[image]

Maybe what's needed is a minimum $fn constraint?  

After inspecting different $fn values it looks like the pointy cap just doesn't actually fit properly to the rest.  At high $fn the discrepancy is small and hard to notice, but it's still there,  so adding the hull() is probably the easiest solution.

closed 2023-01-06 by revarbat