BelfrySCAD/BOSL2 #1028: Wedge:  center=true behavior unexpected...

RAMilewski, 2023-01-18:
...or at least not well explained in the wiki.

```
include <BOSL2/std.scad>
size = [20, 20, 10];
color("PaleGreen") cuboid(size);
up(size.z) color("yellow") wedge(size, center = true);

```
[image]

In this example the "center" of the wedge in the Y direction is some percentage (25% ?) of the Y dimension from the tall end of the wedge.   

What is it I'm misunderstanding?

revarbat, 2023-01-19:
This would be a bug.

revarbat, 2023-01-19:
It appears that the attachment geometry for `wedge()` is trying something clever using `shift=`, which has a side effect of moving the `CENTER` anchors.

revarbat, 2023-03-02:
This bug is also existent in `right_triangle()`

adrianVmariano, 2023-03-16:
Actually....what should the CENTER anchor for a right_triangle or wedge be?  Or for that matter, a trapezoid?  

Right now the FWD anchor of a trapezoid or right triangle is the center of the front edge, and the BACK anchor is the center of the BACK edge.  The CENTER anchor is the midpoint between those two.  This seems reasonable, no?  

One could define it as the centroid, but that seems likely to be confusing.  One could define the center anchor as the point with the x coordinate from the front edge center and the y coordinate midway between the front and back edges.  Is that better?  Then presumably the BACK anchor would be in line with the center, though that means the back anchor for a triangle (and hence wedge) would not be in the shape.  

So what's the right behavior?

RAMilewski, 2023-03-17:
Is it really using front and back edge midpoints now?   It doesn't look like it in the example above.

RAMilewski, 2023-03-17:
Nor here:
...
include <BOSL2/std.scad>
size = [20, 20, 10];
wedge(size, center = true);
...
[image]

adrianVmariano, 2023-03-17:
Yes.  It's a little easier to see what's going on if you stick to 2d, but if you insist, here's your model with show_anchors():

[image]

See now the BOTTOM+RIGHT anchor is visible at the midpoint of the triangles long bottom edge.  The TOP+RIGHT anchor is at the triangles apex.  The RIGHT anchor is midway between them.   

The actual CENTER anchor is just centered on the extrusion.  There's no confusion about what happens in the X direction.  

So again I ask, where SHOULD the anchors be?  What did you expect ?  Why is this wrong?  It may be easier if you play with right_triangle and trapezoid for answering these questions.  (Those modules both behave the same way, since right triangle is basically a degenerate trapezoid.)

RAMilewski, 2023-03-18:
I think my problem was expecting to "center" a stack of objects with identical x dimensions and have them aligned.
It's the utility of the center argument to wedge that I was questioning.  Not so much technically as semantically.

Certainly if you explicitly use attach(), then everything plays as expected:

```
include <BOSL2/std.scad>
size = [20, 20, 10];
recolor("PaleGreen") cuboid(size)
    attach(TOP) recolor("yellow") wedge(size, anchor = BOT);
```
[image]

Using the triangle as a model for understanding the anchors in the wedge raises additional questions;

```
include <BOSL2/std.scad>
size = [20, 20, 10];
left(25) rot([-90, -90, 0]) left_half(x = 0.5) right_half(x = -0.5) wedge(size, anchor = CENTER) show_anchors();
right(20) right_triangle([20,10], anchor = CENTER) show_anchors();
```
[image]

1. Taking a center slice of the wedge and show_anchors() shows three anchors at the upper apex of the wedge slice, whereas the right_triangle has only two.  On the wedge they are TOP, TOP+LEFT, and... ???
2. Is there really no RIGHT anchor point?

Also, the docs for show_anchors() mentions "custom" anchors, but I can find neither a description of them nor information on how to use them anywhere in the wiki.  Can you point me in the right direction?

Thank you for all your kind help as I try to drink from the fire hose that is BOSL2.

adrianVmariano, 2023-03-18:
Degenerate trapezoid showing the BACK+LEFT anchor:

[image]

RAMilewski, 2023-03-18:
I should probably write this up as a separate issue, but....

The wiki says the custom argument to show anchors defaults to true.  But there are no names showing.  Setting the std argument to false and explicitly setting the custom argument to true shows no anchors at all.

adrianVmariano, 2023-03-18:
Did you choose an object where named anchors actually exist?  Most objects don't have them.  They will be listed in the Named Anchors section of the wiki entry.   Try `star(id=30,od=75,n=5) show_anchors();`

It appears that there is special code for trapezoids with a zero length top/bottom edge that defines the anchors in both the LEFT and RIGHT directions to be the bisector of the angle (?) or something like that.  This seems like a weird choice and bad because it's discontinuous---if you make the back edge 1e-4 you get different anchors.  The normal anchors are the half angles between vertical (the BACK direction) and the face on the specified side, which seems reasonable....and still reasonable for the triangle case, as much as anything.)    I guess Revar was trying to make the anchors for the triangle match the anchors for the other two vertices.  So if I add those to the picture:

[image]

A triangle with four anchors, the two BACK ones being coincident.  I changed it and now get this:

[image]

It seems like if we need to support a symmetric anchor for the BACK vertex of a degenerate trapezoid or right triangle it needs to be handled using a named anchor.   I don't know if that's necessary.  To be honest, I never use the anchor directions on corners, just the locations.

adrianVmariano, 2023-03-19:
Revar has proposed in the BOSL2 that right triangle have completely different (and inconsistent) anchors compared to trapezoid.   Presumably this would carry over to wedge if implemented.   He just started from nothing and asked what the anchors should be, and concluded that BACK should be the back corner pointing in Y direction, BACK+RIGHT should be on the hypotenuse pointed perpendicular, RIGHT should be the right hand corner pointed in x direction, and CENTER should also be on the hypotenuse, but pointed in the Y direction.  LEFT+BACK and RIGHT+FWD he wasn't sure about, but maybe they are at corners and point in the half-angle directions.  

Thoughts on this?  Is it more confusing to be different than trapezoid?  Or more confusing to be less intuitive about the triangle?  Is this proposal in fact more intuitive?  In particular, CENTER is no longer in between FRONT and BACK.  Having the center point and the BACK+RIGHT point be the same seems odd.

RAMilewski, 2023-03-19:
I don't think having the right triangle and trapezoid having different anchors is a problem.  If anything, it offers more choices to the designer.  I would vote for LEFT+BACK and RIGHT+FWD pointing in the half-angle directions.   But...

Remember that I'm they guy who got befuddled over cuboid( rounding = ...), prismoid, and rounded_prism, so take my votes with a couple of grains of salt.    

For newcomers to OpenSCAD it takes a little while to grok the multiplicity of ways to get to a given shape.  Making an equilateral triangular prism by starting with cyl( ...,  $fn = 3) comes as a bit of a surprise to rank beginners  (who are still trying to come to terms with immutable variables and the absence of a case statement).

adrianVmariano, 2023-03-19:
I think it's not so important where those "corner" anchors point.  The big question I see is where should the CENTER anchor be?  Is Revar right that putting it in the same place as the RIGHT anchor, but pointing in a different direction, makes sense?  I feel like someone might do `right_triangle(....,anchor=RIGHT)` and `right_triangle(...,anchor=CENTER)` and be really puzzled that nothing changed.  That is, having the RIGHT and CENTER anchor points be the same is weird and unexpected.  

Having the anchors different only offers more choices to the designer if they are actually *different* and not simply relabeled.  For me, I don't think I have ever made use of an anchor direction that wasn't perpendicular to one of the faces/edges of a shape, so I don't even have a use case for those anchor directions.   I also suspect that the CENTER anchor of a triangle may not be useful, given the trouble we are having in deciding where it should be.

RAMilewski, 2023-03-20:
I'm not at all certain that I have a good enough grasp of anchoring in general to opine on this, but I won't let that stop me:

1. The anchors for FWD, BACK, and LEFT are all located at an extreme edge of the triangle and point in the direction of the X or Y axis.
2. The anchors for FWD and LEFT are located at the midpoint of an edge of the triangle.
3. The anchor for BACK is located at the back-most point on the triangle.

Given all of that, for a right_triangle([X`, Y`) I would: 

1.  Put the RIGHT anchor at the right-most point on the triangle pointing X+.
2. Put the BACK+RIGHT anchor at the midpoint of, and normal to,  the hypotenuse.
3. Put the CENTER anchor at [X'/2,Y'/2] pointing BACK.

I'm not sure I understand the direction of the current RIGHT+FWD anchor.  I would expect something closer to a 45 degree angle from the x axis.    ...but as I said I'm a bit foggy on the general case.

adrianVmariano, 2023-03-20:
So you are putting the CENTER anchor on the midpoint of the hypotenuse, so at the same point as the BACK+RIGHT anchor.   Or at least I think that's it?  Not sure what X' and Y' mean.  

Note also that with possible exception of the directions being slightly different, your proposal, like Revar's, is a relabeling, not a fundamental change.  

I think the current RIGHT+FWD anchor points in the direction that is opposite the angle bisector of the triangle.

closed 2023-04-05 by revarbat