BelfrySCAD/BOSL2 #1029: Distributors:  Wiki needs additional example(s).

RAMilewski, 2023-01-19:
In the description of xcopies() and other distributors in the wiki it says:

> Side Effects:
>    $pos is set to the relative centerpoint of each child copy, and can be used to modify each child individually.
 >   $idx is set to the index number of each child being copied.

An example showing how to modify one or more of the children would be quite helpful.  

All of my attempts to modify a child independently have failed.  For instance:

```
include <BOSL2/std.scad>
$fn =36;
cuboid (10, anchor = BOT);
ycopies (10) {
    echo($pos);
    if($idx==1) { $pos = [0,15,0]; }
    color("red") xcyl(l = 10, d=10, anchor = BOT);
}
```
[image]

revarbat, 2023-01-19:
Err, you only changed the value of a variable, in a context that it is never used in. 

Try this instead:
```openscad
include <BOSL2/std.scad>
$fn =36;
cuboid (10, anchor = BOT);
ycopies (10) {
    echo($pos);
    multmatrix($idx==1? back(15): ident(4))
        color("red") xcyl(l = 10, d=10, anchor = BOT);
}
```

RAMilewski, 2023-01-20:
OH!!   OK... Seriously cool!  I'm a bit of an OpenSCAD beginner and missed multmatrix() in learning what do know about OpenSCAD.   
I've not done much matrix math since hand coding Postscript on the original Apple Laser Printer in the late 1980's.   This will be fun!  Thank you!!

I do still think an additional example in the wiki would be helpful.  ...especially for those of us still learning OpenSCAD and the awesomeness of BOSL2.  Matrix math scares lots of people.

revarbat, 2023-01-20:
Keeping this open until said examples are added.

adrianVmariano, 2023-01-31:
I wrote an introduction to add to distributors.scad as a section on adaptive children.  Any comments?  Did I hit everything that needs hitting?  Also, can I use "Continues:" with examples?  

    // Section: Adaptive Children Using `$` Variables
    //   The distributor methods create multiple copies of their children and place them in various ways.  While there are many occasions where
    //   a model demands multiple identical copies of an object, this framework is more powerful than
    //   might be immediately obvious because of `$` variables.  The distributors set `$` variables that the children can use to change their
    //   behavior from one child to the next within a single distributor invocation.  This means the copies need not be identical.  
    //   The {{xcopies()}} module sets `$idx` to the index number
    //   of the copy.  The first example shows how we can use that to produce **different** geometry at each index:
    // Example(2D):
    //   xcopies(n=10, spacing=10)
    //     text(str($idx));
    // Example(2D): Here the children are sometimes squares and sometimes circles as determined by the conditional statement.  
    //   xcopies(n=4, spacing=10)
    //     if($idx%2==0) circle(r=3,$fn=16);
    //     else rect(6);
    // Continues:
    //   Suppose we would like to color odd and even index copies differently.  This example shows two important gotchas.  First of all, the `if` statement
    //   in the previous example works for creating geometry, but don't be tempted to use an `if` statement to set variables like `if (condition) { c="red";}`
    //   because the variable is set only in the scope of the if statement and isn't available later on.  Instead you must use the ternary operator as shown in the example.  
    // Example(2D):
    //   xcopies(n=6, spacing=10){
    //       let(c = $idx % 2 == 0 ? "red" : "green")
    //           color(c) rect(6);
    //   }
    // Continues:
    //   The second complication is that in OpenSCAD version 2021.01 and earlier, assignments in children were executed before their
    //   parent.  This means that `$idx` isn't available in assignments, so you will get a warning about an unknown variable.
    //   Two workarounds exist, neither of which are needed in newer versions of OpenSCAD.  The workarounds solve the problem because
    //   **modules** execute after their parent, so the `$` variables **are** available in modules.  In the example above, `let()` is a module
    //   that sets variables available to its children.  Note that multiple assignments in `let()` are separated by commas, not semicolons.  
    //   The other workaround is to wrap your child in a `union()`.  The next example shows how you can change the position of children
    //   adaptively.  If you want to avoid repeating your code for each case, this requires storing a transformation matrix in a variable
    //   and then applying it using `multmatrix()`.
    // Example(2D):
    //   xcopies(n=5,spacing=10)
    //     union()
    //     {
    //       shiftback = $idx%2==0 ? back(5) : IDENT;
    //       spin = zrot(180*$idx/4);
    //       multmatrix(shiftback*spin) stroke([[-4,0],[4,0]],endcap2="arrow2",width=1/2);
    //     }
    // Continues:
    //   In these examples we used `$idx`, but the various distributors offer a variety of `$` variables that you can use in your
    //   children.  Check the "Side Effects" section for each module to learn what variables that module provides.

adrianVmariano, 2023-02-04:
This is done.

closed 2023-02-04 by adrianVmariano