BelfrySCAD/BOSL2 #1032: text3d() - valign and halign appear inoperative

RAMilewski, 2023-01-22:
The examples show equivalent operations using anchors, but the valign and halign parameters appear to have no effect on text placement.

```
include <BOSL2/std.scad>

text3d("Foobar", h = 3, size = 5, valign = "bottom", halign = "left");
fwd(10) text3d("Foobar", h = 3, size = 5, valign = "center", halign = "right");
fwd(20) text3d("Foobar", h = 3, size = 5, valign = "top", halign = "center");
```

[image]

adrianVmariano, 2023-01-23:
Yeah, they are documented to work, but the code completely ignores them.  Revar will have to figure out whether to make it work or change docs to match code.   Note to Revar: merge the PR #1030 first to ensure no conflicts.

adrianVmariano, 2023-03-07:
I am not sure that text3d() is designed correctly.  The anchor mechanism it provides seems kind of awkward.  The default anchor is "baseline[-1,0,-1]" for example, which is kind of cryptic.  It seems silly to have a Y coordinate that must always be zero...or is ignored?   I think this whole kludge should be scrapped.   I'm wondering if instead there should be maybe `baseline=false` and `baseline=true` that would affect whether the "CENTER" anchors are at the baseline or center in the Y direction.  This would basically be like changing the anchor type, so another option would be to have it be atype="baseline" or atype="center".   Another thing I was thinking is if halign and valign are to be suported then we should add zalign to go with them.   Another idea is that the various align parameters could be extended to accept vectors like LEFT, RIGHT, etc.  Not sure if that's a good or bad idea.  So `halign=LEFT` would work in addition to `halign="left"`.  We could also consider just not supporting the "old style" alignment methods.  

@revarbat Thoughts?  

While I was fiddling around I also noticed that plain openscad text() accepts a valign of "bot" without error but ignores it.  Other bogus values give an error.

adrianVmariano, 2023-03-13:
The decision has been made to axe support for halign and valign.  And I've rewritten anchoring to be a little less convoluted.

closed 2023-03-14 by revarbat