BelfrySCAD/BOSL2 #1036: Breaking changes - Use releases

nick-randal, 2023-01-28:
Releases are not being used and there are regular breaking changes with no clear way to migrate forward. 
Requests
- Please use releases, so we can easily rely on a known version
- Use semantic versioning to indicating, Breaking, Features, Patches
- Provide migration tips for more complicated operations like bezier

I will leave a project for a period of time and come back to a completely broken script.

Fundamental operations like `move` and gets used heavily in scripts was a significant breaking change. However, there is no semantic versioning being used, it is all under the major version of 2.

adrianVmariano, 2023-01-31:
I feel like Revar's attention has been elsewhere lately, but my impression is that it was always his intention to use releases and to have stable versions with well identified version numbers.  The problem is...the first Release hasn't happened yet.   Maybe this Issue will help push the project towards that release happening sooner.   I've been feeling like it was just around the corner for...the past 6 months.

RAMilewski, 2023-02-13:
Reading through the issues and PRs, it looks like a couple of things work differently in versions of openSCAD _newer_ than 2021.01.  I've used some of the development snapshots without any real problems, but I've yet to find a changelog or revised docs for openSCAD, so I do so with some trepidation.    Perhaps some monthly or quarterly  "0.x releases" of BOSL2 would help put a few markers in place to help with this?

The continually updated BOSL2 wiki a great help because it's easier to tell what a bit of code is supposed to do.  Would that openSCAD would move the docs to a wiki instead of the unsearchable wikibooks.  

/rant

pipatron, 2023-07-06:
### Local copy

One simple solution is to keep a dedicated directory for each major 3D project you are working on, and keep a local copy of BOSL2 inside that directory. Since OpenSCAD first looks in the script's local directory it will find the correct BOSL2 files even if you have the latest version in the main library path. It adds 

### git

If you're a developer and/or know how to use git, this enables two major features:
 - You can come back to a project, do a `git fetch` to see if there are any changes, and decide to apply them or not.
 -  You can just upgrade straight away and use `git bisect` to find the exact point where something breaks. This makes it easier to modify the project one breaking change at a time.

For simple one-off OpenSCAD scripts you could just add a short comment with the BOSL2 git hash you're currently using. If you want to use the script in the future and nothing works, you can copy it to a new directory, check out the BOSL2 project with that hash and you should be good to go. If you already have an updated version checked out somewhere else you can just clone it from your local copy - it has the full history.

revarbat, 2023-08-26:
Hmm.  It seems that versions.scad's version number hasn't been bumped by the PR checkin automation since Jun 21, 2021.

revarbat, 2023-08-26:
I can also see having monthly dev snapshot versions auto-packaged, once we fix that version bumping.

DeflateAwning, 2023-10-28:
I support this request. Why are semantic versions and releases not being used in this project?

It's so exctiting to see the new features and progress being made, but I wish there were releases that follow the normal software dev style. This is, by far, the most-stared project I've ever seen that doesn't follow semantic versioning

revarbat, 2026-10-10:
Semantic versioning with weekly release snapshots was implemented some months ago.

closed 2026-10-10 by revarbat