BelfrySCAD/BOSL2 #1038: Tapersample / higsample is not required in spiral_sweep()

dr-occams-r, 2023-01-28:
Tapersample / higsample is not required in `spiral_sweep()`.  It breaks from the coherent use of the fragmentation parameter $fn.  I didn't see it used by any other functions other than `thread_helix()`, so I don't think it will break anything to remove it.

Thank you @adrianVmariano for pointing me to the exact source of this problems (ref issue #741).

As a test, I deactivated the oversampling by simply setting `tapersample = 1` in `spiral_sweep()`. 

The test code is: 
```
include <BOSL2/std.scad>
include <BOSL2/threading.scad>

zdistribute(spacing=5){
  thread_helix(d=10, pitch=2, thread_depth=0.75, flank_angle=15, turns=1, left_handed=true, taper=3, $fn=5);

  thread_helix(d=10, pitch=2, thread_depth=0.75, flank_angle=15, turns=1, left_handed=true, taper=3, $fn=10);

  thread_helix(d=10, pitch=2, thread_depth=0.75, flank_angle=15, turns=1, left_handed=true, taper=3, $fn=15);

  thread_helix(d=10, pitch=2, thread_depth=0.75, flank_angle=15, turns=1, left_handed=true, taper=3, $fn=20);

  thread_helix(d=10, pitch=2, thread_depth=0.75, flank_angle=15, turns=1, left_handed=true, taper=3, $fn=50);

  thread_helix(d=10, pitch=2, thread_depth=0.75, flank_angle=15, turns=1, left_handed=true, taper=3, $fn=100);
}
```

Screenshot below of the result show mesh complexities that are coherent with the specified fragmentation $fn.  
 
[image]

Thus, based on the results of this test, I would suggest completely removing the "Tapersample / higsample" logic.   I did not submit a PR because the team is actively working on these files and it's will just be more trouble for everyone if I add a PR.  There is no rush for this, it can be done next time the relevant code is changed.  

Please provide feedback on any counter arguments to this, or any side effects that I might not have considered.

adrianVmariano, 2023-01-28:
Removing this requires the user to increase the fragments by a factor of 10 to get the same rounded end, and that potentially propagates through the whole model, forcing the entire model to a higher number of fragments.   If we were to make this change we should presumably consider changing the tapering function, as I don't think it can be well expressed without the higher sample rate. 

You complain about incoherence, but what does $fn mean?  If you make a big circle and little circle and set $fn=100 both have 100 fragments.  In the threading, the radius of the rounded end is much smaller than the radius of the whole thread diameter, it's something on the scale of the thread width.  So to put the right number of fragments in at the tapering tip, I need the fragments to be much smaller, proportional to the thread width rather than thread diameter.  I would say that the increased sample rate there is an effort to enforce *consistency*, that you get the detail you asked for, and not an overly coarse representation of that taper.  The effective radius is smaller, so the fragments are smaller.   Ideally the fragment spacing would change adaptively so you don't have so many during the flatter portion of the higbee, but you have more towards the tip.   That kind of adaptive change seemed overly complex.  

I guess we can see what @revarbat thinks about it.

adrianVmariano, 2023-01-28:
Another point: I think the main reason for thread_helix to exist is that oversampled taper.  If we get rid of that we should probably consider getting rid of thread_helix entirely, unless another argument can be brought for its existence.   I need to do some testing to figure out the relative performance vs threaded_rod so I have solid data on which to base decisions.

adrianVmariano, 2023-03-02:
I did some tests using the thread example from #741 and it appears that with that example, the thread feeds more smoothly with the 10x oversampling.   The print with oversampling self-feeds without stopping reliably, whereas the one without oversampling will stop feeding, and doesn't seem to turn as freely.  So on the basis of this observations, I'm not inclined to eliminate the oversampling.  

It does seem reasonable, though, to expose the oversampling parameter to the user.  So once the dust settles on the naming conventions for thread ends I'll do that.

adrianVmariano, 2023-04-19:
This has been done.

closed 2023-04-19 by adrianVmariano