BelfrySCAD/BOSL2 #1039: [BUG] Cannot create M1.4 screw with a head

LenStruttmann, 2023-01-30:
This works:
```
screw("M1.4,5", head="none", thread="none");
```

This does not work:
```
screw("M1.4,5", head="pan", thread="none");
```

Log:
```
Compiling design (CSG Tree generation)...
[ERROR: Assertion 'is_def(entry)' failed: "Screw size M1.4 unsupported for headless screws" in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad, line 2648](https://github.com/revarbat/BOSL2/issues/2648,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[TRACE: called by '_screw_info_metric' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad, line 1789](https://github.com/revarbat/BOSL2/issues/1789,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[TRACE: called by 'screw_info' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad, line 468](https://github.com/revarbat/BOSL2/issues/468,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[TRACE: called by '_get_spec' in file C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad, line 524](https://github.com/revarbat/BOSL2/issues/524,C:/Users/busin/Documents/OpenSCAD/libraries/BOSL2/screws.scad) 
[TRACE: called by 'screw' in file HC-SR04UltrasonicModule.scad, line 186](https://github.com/revarbat/BOSL2/issues/186,D:/OpenSCAD/HC-SR04UltrasonicModule.scad) 
[TRACE: called by 'translate' in file HC-SR04UltrasonicModule.scad, line 185](https://github.com/revarbat/BOSL2/issues/185,D:/OpenSCAD/HC-SR04UltrasonicModule.scad) 
[TRACE: called by 'for' in file HC-SR04UltrasonicModule.scad, line 183](https://github.com/revarbat/BOSL2/issues/183,D:/OpenSCAD/HC-SR04UltrasonicModule.scad) 
[TRACE: called by 'HCSR04screws' in file HC-SR04UltrasonicModule.scad, line 20](https://github.com/revarbat/BOSL2/issues/20,D:/OpenSCAD/HC-SR04UltrasonicModule.scad) 
[TRACE: called by 'if' in file HC-SR04UltrasonicModule.scad, line 14](https://github.com/revarbat/BOSL2/issues/14,D:/OpenSCAD/HC-SR04UltrasonicModule.scad) 
```

**Expected behavior**
Be able to model an M1.4,5 screw with pan head.

- OpenSCAD Version: 2021.01

adrianVmariano, 2023-01-30:
There is a bug, but not what you thought.  The bug is that the error message is bad.  It should tell you that the head you requested is an unsupported size.  I can't make heads (except cheese?) on an M1.4 screw because the ISO tables don't list a size for such a head, no no info on how big to make it.   So once it's fixed it will say "Screw size M1.4 unsupported for pan head screws".

LenStruttmann, 2023-01-30:
Yeah, I figured that would be the case.    Thanks!

adrianVmariano, 2023-01-30:
If you find a source that gives more head data we can add it.

adrianVmariano, 2023-01-30:
But if you need that head at the moment you'll need to follow the method in Example 7 where you set the head parameters in the screw structure.

LenStruttmann, 2023-01-30:
https://www.metricscrews.us/index.php?main_page=product_info&cPath=13_227_230&products_id=930

Head Diameter: 2.60
Head Height: 1.00

But it's no big deal.  As you say, there are work-arounds.  It would be nice, however, to get the error message fixed.  Thanks!

adrianVmariano, 2023-01-30:
So the problem with that reference is we can't tell if they're selling some oddball screws that they fabricate to their quirky specs, or if that's some sort of standard.  The fact that they list the driver as JIS is interesting, because JIS doesn't really exist any more.  (The standard was discontinued several years ago and replaced by the equivalent ISO, which doesn't have the same geometry that the defunct JIS did.)

I have a fix for the error message implemented.

closed 2023-01-30 by adrianVmariano