BelfrySCAD/BOSL2 #1040: [BUG] spiral_sweep ignores `turns` for left-handed spirals

jon-gilbert, 2023-02-01:
**Describe the bug**
Some time in the past 4 months `spiral_sweep()` stopped producing left-hand spirals with a negative `turns` argument. 
skin.scad's `spiral_sweep()` details this in its documentation as how to create a left-handed spiral:
`//   If turns is positive the path will be right-handed;  if turns is negative the path will be left-handed.`

**Code To Reproduce Bug**
```
// pulling directly from spiral_sweep's Example 1:
include <BOSL2/std.scad>
poly = [[-10,0], [-3,-5], [3,-5], [10,0], [0,-30]];
xdistribute(spacing=50) {
    spiral_sweep(poly, h=200, r=50, turns=3,  $fn=36);
    spiral_sweep(poly, h=200, r=50, turns=-3, $fn=36);
}
```
...which yields the following model:
[image]

**Expected behavior**
In the example above, I'd expect to have two spirals moving in opposite directions. 

**Additional context**
- OpenSCAD Version: 2021.01 on Windows 10 Home (22H2) 19045.2546 
- Other libraries used: BOSL2 is from whatever BOSL2-master.zip represented on or about 2023/01/27

adrianVmariano, 2023-02-01:
Hmmm.  Weird.  This seems to have been the interaction of a bug where a negative height caused the spiral direction to flip, which caused me to "fix" spiral_sweep because I didn't know the negative heighit was happening.  So I think I have a fix for this.

jon-gilbert, 2023-02-03:
Agreed, most recent BOSL2-master.zip repairs this. Thanks!