BelfrySCAD/BOSL2 #1050: [BUG]

R512-0, 2023-02-06:
**Describe the bug**
Hi everyone,
i found some strange behavior or is it intended to be like this?:
1. if i first create a screw_info (see code below):
    threads_oversize in screw_info changes the size of the screw_hole,
    but a diferent amount than hole_oversize as direct parameter does. 
    It also changes the size of the head, but it is called THREAD_oversize?
2. screw_hole() doc:
    the parameter is:       hole_oversize
    it is called in the doc: oversize_hole
3. is there a simple way to enable threads, when creating a hole from a screw_info?
    This is my ugly solution:
    screw_hole(screw_info,length=10,thread=struct_val(screw_info,"pitch"));

**Code To Reproduce Bug**
```
// code for 1

include <BOSL2/std.scad>
include <BOSL2/screws.scad>

$fn=60;
screw_info = screw_info("M5",head="socket",threads_oversize=2);
xdistribute(15){
    screw_hole(screw_info,length=10);
    screw_hole("M5",head="socket",length=10,hole_oversize=2);
}
```

**Expected behavior**
1. shoud be the same behavior in both cases?
2. should be equal in doc
3. maybe should be thread=true

adrianVmariano, 2023-02-06:
I think you should be able to say `thread=true` to get a threaded hole for #3.  I have fixed this so it works in my dev fork.

For #2, I have fixed the doc error.  

For #1, I'm not sure what's going on.  I originally created the screw specs to be an internal storage for the screw data, and then later made them user accessible to give users more options, but I never though they were going to be the primary way of using screws.   I haven't done much testing of using them the way you are trying to.  I think that the oversize parameters are supposed to vanish if the structure gets inherited, so that if you make a screw as a child of a hole the screw has the right size.  (The oversize parameter of a normal screw hole is set to the normal amount that holes are enlarged by.)   But there are problems.  I also noticed that setting head_oversized caused the hole for the head to get smaller instead of larger.  So some further investigation is going to be needed to figure this out.  

So screw_hole normally makes the hole according to to the tolerance you give, which specifies the standard amount to oversize the hole.  If you give hole_oversize to screw_hole you're asking to **further** enlarge the hole from that amount.

adrianVmariano, 2023-02-17:
I think I have fixed #1 in the PR.  Note that I changed threads_oversize to shaft_oversize to be more accurate, since it also affects non-threaded parts of the shaft.

adrianVmariano, 2023-02-18:
The PR with the fixes has been merged.

closed 2023-02-18 by adrianVmariano