BelfrySCAD/BOSL2 #1054: [BUG] rect_tube(irounding) default fails

RAMilewski, 2023-02-13:
**Describe the bug**
The wiki says 
 _
irounding | The roundover radius for the inside edges of the rectangular tube. Default: Same as rounding
_
However, irounding appears to default to zero.

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>

corner = 5;
tube_size = [30, 20];
height = 5;

left(tube_size.x/2 + 5)
    rect_tube (size = tube_size,  wall = 2, h = height, rounding = corner);

right(tube_size.x/2 + 5)
    rect_tube (size = tube_size,  wall = 2, h = height, rounding = corner, irounding = corner);

```

**Expected behavior**
Inside of tube should have rounded corners when rounding is defined and›ﬁ irounding is not specified.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2021.01
- Other libraries used: None.

RAMilewski, 2023-02-17:
And just to muddy the waters a bit...

If irounding defaults to rounding, the wall thickness in the corners exceeds the value of the wall argument.

Should it perhaps default to rounding - wall?     Or is that just an over-complication?

```
include <BOSL2/std.scad>

corner = 5;
tube_size = [30, 20];
height = 5;
wall = 2;

left(tube_size.x/2 + 5)
    rect_tube (size = tube_size,  wall = wall, h = height, 
        rounding = corner, irounding = corner);

right(tube_size.x/2 + 5)
    rect_tube (size = tube_size,  wall = wall, h = height, 
        rounding = corner, irounding = corner - wall);

```
[image]

adrianVmariano, 2023-02-17:
Heh.  I came here to point about this very matter.  It seems like the documented default is wrong, but I'm a little uncertain about the best behavior.  Your proposal above isn't right either, though.  Consider the case of corner=0.  Or corner<wall.  Should it be max(0,corner-wall)?  I like the idea of choosing a default to maintain a uniform wall thickness.  That makes the default useful.

RAMilewski, 2023-02-17:
Oh... You're right!   I didn't think that through.  max(0,corner-wall) makes more sense.    Both you and Revarbat have orders of magnitude more experience here than I.   I think my only value to the project is just being a  "bear of little brain" who is seeing much of this for the first time, and trying many of the examples as written.

(...apologies to A.A. Milne)

adrianVmariano, 2023-02-17:
Yeah, well, there is a **ton** of stuff in BOSL2 and just trying stuff out and finding places where things are broken is a valuable contribution.

adrianVmariano, 2023-02-17:
What happens when wall thickness is different on the sides.  

```
   rect_tube(
      size=[100,100], isize=[75,90], h=30,
      rounding=[0,5,10,15]
   );
```

[image]

[image]

Left side is rounding relative to thinner wall; right side is rounding relative to thicker wall.  Guess one could also do the average.  

[image]

adrianVmariano, 2023-02-17:
It seems like rounding should be either relative to the average wall thickness or the min wall thickness...but not sure which is best.

adrianVmariano, 2023-02-17:
Hmmm.  Wall thickness can differ at the ends.

adrianVmariano, 2023-02-18:
I think I have this fixed in the PR.

adrianVmariano, 2023-02-18:
PR merged so this should be fixed.

closed 2023-02-18 by adrianVmariano