BelfrySCAD/BOSL2 #1057: Variable width along path for offset_stroke or implement stroke as a function

alidaf, 2023-02-21:
This feature is available for stroke but since stroke is a module only, it can't be used with functions/modules such as skin. 
Offset_stroke can be used with skin but currently lacks the ability to vary width along the path length.

It would be incredibly useful for either stroke to return a closed path as a function or for offset_stroke to incorporate the variable width in the same way as stroke.

alidaf, 2023-02-21:
Thanks, although I don't know if I'm just missing a trick with the skin function or if there is an alternative. Essentially I want to extrude a collection of logarithmic spirals where the logarithm parameters as well as rotations are modified as a function of height. My current method is to use skin in combination with the offset_stroke where the various parameters are modified by Bezier curves as a function of height. It works well but I can generate more interesting 2D shapes by using stroke with the variable width. Extruding and scaling a stroked path at different heights also scales the stroke line width, which is what I don't want to happen. This approach also kills the CPU. Skin seems like a much more elegant method but only takes regions/paths.

adrianVmariano, 2023-02-21:
Not sure I fully understand what you're doing, it it sounds like you're applying offset_stroke to a logarithmic spiral with intent to then skin it with other such spirals?  And it sounds like you're doing this for a kind of artistic purpose, not because you need an exact match to some specific spiral with certain tapering properties.  

So in that case, it seems like the simplest thing is to simply add your own angle-dependent offset to the spiral.  That is, do the offset operation yourself.  And you can simplify things by just doing it in the radial direction rather than bothering to do it actually in the normal direction.  I think it will be close enough.  Then you can vary your width according to whatever functional form you desire.  Note that if you really do want to do it in the true normal direction, then just compute the normal to the logarithmic spiral by taking derivatives and use that to do your offset.  Because your data is continuous you don't need the complicated part of the offset function that deals with self-intersection of the offset path.  Or even the part that deals with adding roundings or chamfers---that latter part seems like it might be complicated somewhat by variable offsets.

alidaf, 2023-02-21:
I think you got the gist of it and I also think I got the gist of your suggestion. It is an artistic endeavour but I'm an autistic engineer with OCD so accuracy is important to me. The point is that it will be a parameterised way of creating organic shapes simply by changing variables. It'll be more of a mathematical challenge than I'm used to but worth a shot. I wonder whether I can somehow join a spiral with it's slightly different twin (marginally different growth factor) and then sort out the ends with circles. That should allow me to get spirals that increase in thickness, but not the other way! How do I fill it? Polygon? Perhaps a better option would be to hull circles along the spiral path, maybe with hull2d_path()? I think I'll do some scoping tests for the latter. Cheers.

adrianVmariano, 2023-02-21:
So in a context like this, only you can decide what ultimately is satisfying and meets your artistic vision.  But I think that the simpler idea I proposed is a strong contender.  If you were talking about a uniformly thick stroke, then this idea wouldn't work because the stroke width will vary a little bit, and you won't have direct control over the stroke thickness.  But since you want to vary the stroke width any way, the fact that it will have intrinsic variation, or that you can't directly control its width anywhere could be acceptable.  In any case, if it's not, the fully accurate approach is mathematically tractable----just more complicated.  You'll need to remember your calculus in that case.  

My guess is that joining a spiral with a different growth factor spiral is not going to work well unless they are very short, because the different will grow huge.  Assuming the difference isn't too big, path_join() might be helpful for adding a curved end, but I think you'll have to compute the curved end yourself to ensure that you don't lose points, which would cause problems for skin.   (And also because you can't join parallel lines with path_join----the extensions of the curves have to meet somewhere.)  

A hull of circles is basically like using stroke().  It builds up the path with a sequence of circles and trapezoids that join the circles, which is essentially the same thing.  So it would work the same way, presumably, with the same limitations.   It would need to be a chain hull, not a hull of all the circles, or you'd fill in the middle of your spiral.  There is no functional form that does a chain hull.

adrianVmariano, 2023-02-21:
Just to add another note, "accuracy" only means something with reference to a target goal.  If your target goal is that you want the normal width of the curve to be specified, then achieving accuracy will require computing the normal direction.  But the simpler functional form is mathematically exact and hence "accurate" in its own way, for the thing it describes.

alidaf, 2023-02-21:
Yes, I'm looking at your way. The polar slope is constant so I'm writing functions to return coordinates of an offset from points along the spiral. Then I think it will be a case of tracking along the negative side until the end, get the points of a semi-circle, track the positive side and then the semi-circle at the start, all in a single path.

adrianVmariano, 2023-02-21:
There is actually also another way to do the offsetting, which would be to use a numerically computed normal vector.  That's actually going to be what happens in the offset() function.   It just wasn't the first thing to pop into my head because it's less accurate.  :)  But if you really want to specifically control the width you could do that.  Compute the normal with path_normal() and then use that to make an offset step.  As long as you keep your point samples far enough apart that the offset path doesn't cross itself, everything will work out ok.  

Note: when I computed the slope it was not constant.

alidaf, 2023-02-21:
I didn't know there was a path_normal()! That may make things easier. I've managed to get a parallel curve but it seems offset. I'm using archimedian spirals, i.e. r = a.e^(k.theta) so the polar slope should be constant. Thanks for the tip.

adrianVmariano, 2023-02-21:
Thing is, path_normal computes numerically and may give a funny result for the end points.  Since you have a functional form you can compute the exact normal mathematically instead of using a numerical approximation.   

The "polar slope", which I assume is dr/dtheta, is not (directly) relevant to the job of offsetting, which happens in cartesian space.  Another way to put it is that you don't need the slope but the tangent vector.  You can compute that from polar coordinates, but I think you'll need a jacobian operator to account for the polar coordinate representation.   Things are messy if you say in polar coordinates, because the unit vectors change.   It will be probably more straight forward to just convert to cartesian and work directly on that representation.

alidaf, 2023-02-21:
Yes, I've found that things go awry at small angles when using the polar slope.

adrianVmariano, 2023-02-28:
Made a new more clear issue to cover this for possible future assessment.

closed 2023-02-28 by adrianVmariano